souliss/souliss #116: assign an lan-ip on souliss 192.168.x.x with decimal

poweredbyant, 2015-03-20:
require 192.168.0.x 
not 192.168.1.x
quick explain and not understanding.. 
ping reponds on pc ... but.also not show up on router page "active lan list"... on the superhub router (virgin media provider), (netgear brand), (UK)
quickcfg.h are ..
uint8_t DEFAULT_BASEIPADDRESS[] = {192, 168, 0, 0};
uint8_t DEFAULT_SUBMASK[] = {255, 255, 255, 0};
uint8_t DEFAULT_GATEWAY[] = {192, 168, 0, 1};

from sketchs
// network addresses
#define network_address_1	0x0059     //192.168.0.89
#define network_my_subnet	0xFF00    // is 225 etc
#define network_my_supern	0x0000    // is 192.168.0.1 (incorrect?)
not pinging on android only on wndows pc ?

thanks Ant

plinioseniore, 2015-03-20:
Hi Ant,

your network setup looks right and if you can ping your device that's good.
In order to get it connected with the Android application you should also
set the Gateway option active.

The node doesn't appear in your router list because probably is shown only
devices that got a DHCP address.

Please join the mailing list and continue the discussion there, the issues
are to list bugs.

Here the link
https://github.com/souliss/souliss/wiki/Community

Regards,
Dario.

From Mobile.

poweredbyant, 2015-03-21:
got it lol it on the quickcfg.h 

0x01		HTTP Command Parser

then it got all working now 
thanks now able looking into it thanks Dario

closed 2015-03-21 by plinioseniore