souliss/souliss #125: more than 37 slots freeze souliss

ribico, 2015-05-05:
By defining the following (or a higher SLOT count) souliss freezes:

#define MaCaco_NODESIZE_INSKETCH
#define MaCaco_NODES			30
#define MaCaco_SLOT				38 // <---- need to be 37 or les to work fine

MaCaco_NODES : 30
MaCaco_SLOT : 37
MaCaco_MEMMAP : FE

MaCaco_NODES : 30
MaCaco_SLOT : 38
MaCaco_MEMMAP : 102

MaCaco_NODES : 30
MaCaco_SLOT : 39
MaCaco_MEMMAP : 106

plinioseniore, 2015-05-09:
Hi Gabriele,

yes there are many cases where the size of the memory map is compared with a byte index, this let the MCU hang and is the reason why you cannot increase the size of your nodes.

I will have a look in detail while working on the branch for the ESP8266, because also in that case I've to fix some issues that were ok only with 8bit MCU.

Hope to have a patch soon.

Thanks,
Dario.

plinioseniore, 2015-05-23:
Hi Gabriele,

would you mind to verify if the node hangs only when the number of slots exceed that value and at same time also the Persistence mode is active?

Dario.

plinioseniore, 2015-05-23:
Gabriele, please try commit 33f91ae and let me know if it fix your problem. I've tried to go though all the possible U8 overflowing in case of large number of slots, but I'm not sure that I've catch them all.

Consider that this commit has also a merge of friariello-porting for 32bit  MCU and ESP8266.

Autohome2, 2015-05-27:
I have been working through my changeover to 7.0.3 
I have downloaded the package and not setup got as only running xp
I get a hang on the gateway over 44 slots with 20 nodes selected?
I have the vnet figure set at 66 not 46 as I need 60 slots?
Thanks
Darren

plinioseniore, 2015-05-27:
Please try 7ad9cba7 and let me know. I assume that this setup wasn't working also in the previos A6.x release.

https://github.com/souliss/souliss/tree/7ad9cba79993f08caea68ff5c58805d12cefc47b
https://github.com/souliss/souliss/archive/7ad9cba79993f08caea68ff5c58805d12cefc47b.zip

Autohome2, 2015-05-27:
Hi will update and check , this combination of slot vs node was working in 614?

Darren

Autohome2, 2015-05-27:
I have updated and tried gateway . with set as 20 nodes and 60 slots and vnet payload at 66 all seems OK
Nice one Dario :-)
Thanks
Darren

Autohome2, 2015-05-30:
since my last comment i have set up git on the new laptop so it has a different version to what i downloaded before, has anything been changed because  the 44 slot max or else the gateway hangs is back again?
Darren

plinioseniore, 2015-05-30:
Hi, which commit were you running before? Nothing has changed since your liste comment on this issue.

Let me know.

plinioseniore, 2015-06-06:
Solved as discussed here
https://goo.gl/pSwGu8

commit 33f91ae5 (2015-05-23): Trying to fix #125
closed 2015-06-06 by plinioseniore