souliss/souliss #146: Input and output states of T2n

plinioseniore, 2015-06-21:
The values of Souliss_T2n_LimSwitch_Close/Open was same as Souliss_T2n_ToggleCmd and Souliss_T2n_OpenCmd_Local. As the close and open state from limit switches was used as either an input and output, this create an incosistency in the logic.

With commit #52c60c8 this has been fixed, but want to have a review by people that have worked out the T2n logic.

ribico, 2015-06-21:
I teste my configuration (without use of LimitSwitches) and it work fine.
I tried to activate the use of limit switches but I could not get it working as expecting... but I assume there's something I don't understand.

From the code comment I found:

		Hardware Limit Switches :
		
			Using bistable switches for identification of open and close position			
				#define Souliss_T2n_LimSwitch_Close	0x14
				#define Souliss_T2n_LimSwitch_Open	0x16

My limit switches come from a 3 way valve with integrated limit switches (similar to a bistable switch)

So I wrote the following in my code:

	if (!digitalRead(MAIN_3WAY_VALVE_BOILER_LIMIT_PIN))
		SetInput(MAIN_3WAY_VALVE, Souliss_T2n_LimSwitch_Open);

	if (!digitalRead(MAIN_3WAY_VALVE_COLLECTOR_LIMIT_PIN))
		SetInput(MAIN_3WAY_VALVE, Souliss_T2n_LimSwitch_Close);

I could not open or close the valve, I think because the logic is always executing the following code portion starting from a limit position.

	else if((memory_map[MaCaco_IN_s + slot] == Souliss_T2n_LimSwitch_Open) || 
			((memory_map[MaCaco_OUT_s + slot] == Souliss_T2n_Coil_Open) && 
				(memory_map[MaCaco_AUXIN_s + slot] == Souliss_T2n_Timer_Off)))
	{
		memory_map[MaCaco_OUT_s + slot] = Souliss_T2n_State_Open;
		memory_map[MaCaco_IN_s + slot] = Souliss_T2n_RstCmd;
  		i_trigger = Souliss_TRIGGED;	
	}

Then I tried the following:

	if (!digitalRead(MAIN_3WAY_VALVE_BOILER_LIMIT_PIN))
             if(mOutput(MAIN_3WAY_VALVE) != Souliss_T2n_State_Open)
			SetInput(MAIN_3WAY_VALVE, Souliss_T2n_LimSwitch_Open);

	if (!digitalRead(MAIN_3WAY_VALVE_COLLECTOR_LIMIT_PIN))
		if(mOutput(MAIN_3WAY_VALVE) != Souliss_T2n_State_Close)
			SetInput(MAIN_3WAY_VALVE, Souliss_T2n_LimSwitch_Close);

But even this does not help.

So, I could not use it.. but maybe because I do not understand the way it should be used.

plinioseniore, 2015-06-21:
If it enter here
```
else if((memory_map[MaCaco_IN_s + slot] == Souliss_T2n_LimSwitch_Open) || 
        ((memory_map[MaCaco_OUT_s + slot] == Souliss_T2n_Coil_Open) && 
            (memory_map[MaCaco_AUXIN_s + slot] == Souliss_T2n_Timer_Off)))
{
    memory_map[MaCaco_OUT_s + slot] = Souliss_T2n_State_Open;
    memory_map[MaCaco_IN_s + slot] = Souliss_T2n_RstCmd;
    i_trigger = Souliss_TRIGGED;    
}
```

The output state changes and it will not longer enter in.

The limit switch just is an input to define the status as closed or open before the end of the relevant timer, otherwise the output state goes in closed/open after a certain time is expired.

ribico, 2015-06-25:
I found a couple of bugs and created a pull request #148.
I tested it with and without limit switches, seems to work fine.

closed 2015-06-25 by ribico