souliss/souliss #149: ESP8266 version won't compile

RussMathis, 2015-06-26:
@anybody, testing souliss--but wont compile gives following error:
file not found: No such file or directory
Error compiling
The erronous code?: #include "WiFiUdp.h"
The file vNetDriver_eth.h, seems to be point at the wrong path for "WiFiUdp.h" but if is change it to <ESP8266WiFi\src\WiFiUpd.>, I get another error: include\slist.h: no such file or directory.

juanpintom, 2015-06-26:
Can you try with the First Steps from this Wiki?? 
https://github.com/juanpintom/Souliss_ESP_Examples/wiki

plinioseniore, 2015-06-26:
You have to add the following line in your sketch
#include <ESP8266WiFi.h>

RussMathis, 2015-06-30:
Still no success,
I'm using windows 7, Arduino IDE 1.6.5, targeting "Generic ESP8266" and followed the instructions from https://github.com/juanpintom/Souliss_ESP_Examples/wiki.

Notes: even after installing souliss library I can not find the example: e01_Hello_ESP8266.ino in the arduino sample sketches. There are sever sketches for souliss but not one by that name.

I have the latest ESP8266 IDE package and it complies sketches OK.

The error I'm getting is:
ESP_Souliss.ino:13:85: fatal error: bconf/MCU_ESP8266.h: No such file or directory
compilation terminated.
Error compiling.

Since I could not find the sample sketch I copied it from the webpage:https://github.com/souliss/souliss/blob/friariello/examples/WiFi/e01_Hello_ESP8266/e01_Hello_ESP8266.ino
and pasted it in Arduino IDE 1.6.5.

Thanks for the help!

juanpintom, 2015-06-30:
Maybe you have the Souliss library on the wrong folder, can you take a look to your library folder? You must have a Souliss folder on it.

Regards

plinioseniore, 2015-06-30:
Looks that you haven't the library properly installed, remove any content
from Documents/Arduino/libraries and follow this guide.
https://github.com/souliss/souliss/wiki/Your%20First%20Upload

Dario.

From Mobile.

closed 2015-08-15 by plinioseniore

bluman74, 2016-03-31:
I'm seeing a similar error on Arduino IDE 1.6.8 when trying to flash the esp8266-01.
I've installed the libraries ESP8266 1.0.0 and Souliss 7.1.1 via the Library Manager.
The library files are located under /home/<user>/Arduino/libraries/souliss.

 Here is the error.

Arduino: 1.6.8 (Linux), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

/tmp/arduino_modified_sketch_562335/e01_Hello_ESP8266.ino:13:85: fatal error: bconf/MCU_ESP8266.h: No such file or directory
 #include "bconf/MCU_ESP8266.h"              // Load the code directly on the ESP8266
                                                                                     ^
compilation terminated.
exit status 1
Error compiling for board Generic ESP8266 Module.

Has anyone else had this issue and resolved it?

plinioseniore, 2016-03-31:
Actually the 7.1.1 compiles on IDE 1.6.5, you have to downgrade to that IDE
and use the supported release of the ESP cores.

Otherwise you need to download manually the latest branch to compile with
that IDE. Please be aware that ESP core 2 are still not supported in any
release of Souliss.

dyezepchik, 2016-04-18:
Have the same issue.
Downloading the latest code didn't help.
But in IDE 1.6.5 sketch got compiled.
Thank you!

plinioseniore, 2016-04-19:
If you want to use espcore 2 and IDE 1.6.8 don't use Souliss 7.1.1 but
download abd install manually the code from branch 720_rc that is on github.

From Mobile.

dyezepchik, 2016-04-19:
I have tried exactly that, downloaded current code from github. But with no success, it didn't compile in IDE 1.6.8. I'll try once again today =)

plinioseniore, 2016-04-19:
You should download the code in 720_rc branch and not the default one.

dyezepchik, 2016-04-20:
oh, thanks

soyxan, 2016-05-13:
I am facing the same problem and I tried the following steps with no luck:

- Uninstalled IDE 1.6.9
- Installed IDE 1.6.8
- Installed Souliss 7.1.1 from the Library Manager
- Installed the ESP8266 board

And I get the same error while trying to compile the example:

```
C:\Users\Vicente\Documents\Arduino\libraries\souliss\examples\WiFi\e01_Hello_ESP8266\e01_Hello_ESP8266.ino:13:85: fatal error: bconf/MCU_ESP8266.h: No such file or directory

 #include "bconf/MCU_ESP8266.h"              // Load the code directly on the ESP8266

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.
```

fastriker, 2016-09-29:
I got the "bconf/MCU_ESP8266.h: No such file or directory" message too. By switching to the 720_rc branch there was no effect. My brother gave me the hint to crate a empty header file in the souliss directory and import it on top of the sketch. With this work around everything works fine.

plinioseniore, 2016-09-29:
The examples in 720_rc have an #include "SoulissFramework.h" that does the trick, if you move to 720_rc but you still use the examples from the previous release, you will not get a compiled code.

jkutianski, 2016-09-29:
@fastriker @plinioseniore 
Is better if you `#include <SoulissFramework.h>` instead of `#include "SoulissFramework.h"`