souliss/souliss #156: Setup Gateway or Peer mode on Runtime

juanpintom, 2015-07-04:
I need to configure if a node works like a Gateway or Peer on Runtime, this combined with the new Wifi-Manager library bring us the possiblity to Setup a new node on a WebInterface.

plinioseniore, 2015-08-18:
@juanpintom @lesjaw

The branch **runtimegateway** is a try to get the requested behaviour. Is available at this link (https://github.com/souliss/souliss/tree/runtimegateway) and need a deep test

So you have now:
SetAsGateway(address)									
RemoveAsGateway()
GetAsGateway()	

As usual, a Gateway shall use SetAsGateway(address) and this will now work at runtime, so a node defined as Gateway in the configuration (conf/Gateway.h) but without this call will no longer act as a Gateway.
This is similar to what was the previous behavior, but now whatever is Gateway related goes under this check.

If you use RemoveAsGateway() the node behave as a Peer at runtime and GetAsGateway returns the node address if is acting as a Gateway, otherwise will return zero.

From the configuration point of view, has been included the configuration file conf/RuntimeGateway.h that is similar to conf/Gateway.h but doesn't set the node as SUPERNODE by default. This is strictly suggested on Ethernet/Wifi only nodes, nodes that bridge other communication media (like USART or wireless) shall add conf/vNetSupernode.h togheter with RuntimeGateway.

Please have a test and report your feedback.

juanpintom, 2015-08-18:
Thank You so much Dario!, You're awesome, I'll try soon as possible an let you know.

You're one step forward than us ;) We're looking to configure a node as GW at start (setup) why we reset the node automatically (software) after change configuration, but you have accomplished this on runtime!!, so we don't need to reset the node now!

Really thanks

lesjaw, 2015-08-18:
awesomeeeee...

i am testing it now.. OMG Dario.. you are such an amazing person..

lesjaw, 2015-08-18:
Here is the vNet and MoCaco debug..

it works!.. no more network flood..

Peer first boot and vNet address request to gateway

.
.
.
.
.
Already connect to : Bolt
Admin Time Out
STA mode started
192.168.0.101
Souliss update IP Address
(vNet)<SETADDR><|0x65>
(IP)<C0,A8,0,65,>
(vNet)<SETADDR><|0x65>
(IP)<C0,A8,0,65,>
HTTP server started
0
Peer Mode
(vNet)<SETADDR><|0xAB00>
(vNet)<SETADDR><|0x65>
(IP)<C0,A8,0,65,>
(vNet)<SETADDR><|0xAB02>
(MaCaco)<OUT><|0x2C|0x0|0x0|0x0|0x0|0x>
(vNet)<BRD><Media><|0x1><|0xB|0x17|0xFFFF|0x65>
(vNet)<BRD><Media><|0x3><|0xB|0x17|0xFFFF|0xAB02>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB02|0xAB01|0x5|0x0|0x0|0x57|0x18>
(MaCaco)<IN><|0x5|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x15|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB01><|0x23|0x17|0xAB01|0xAB02>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB02|0xAB01|0x5|0x0|0x0|0x57|0x18>
(MaCaco)<IN><|0x5|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x15|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB01><|0x23|0x17|0xAB01|0xAB02>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB02|0xAB01|0x22|0x0|0x0|0x3F|0x18>
(MaCaco)<IN><|0x22|0x0|0x0|0x3F|0x18|0x>
(MaCaco)<OUT><|0x32|0x0|0x0|0x3F|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB01><|0x23|0x17|0xAB01|0xAB02>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB02|0xAB01|0x5|0x0|0x0|0x57|0x18>
(MaCaco)<IN><|0x5|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x15|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB01><|0x23|0x17|0xAB01|0xAB02>

ON relay

(vNet)<IN>(0xAB01)<|0xC|0x17|0xAB02|0xAB01|0x14|0x0|0x0|0x27|0x1|0x2>
(MaCaco)<IN><|0x14|0x0|0x0|0x27|0x1|0x>
UDP/AutoRemote connected
GET data to URL: /sendmessage?message=Node;0;Bawah
closing connection
0
No API at all, so Skip!
(MaCaco)<OUT><|0x15|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB01><|0x23|0x17|0xAB01|0xAB02>

Multicast OFF relay

(vNet)<IN>(0xAB01)<|0xC|0x17|0xFFFF|0xAB01|0x34|0x0|0x0|0x11|0x1|0x4>
(MaCaco)<IN><|0x34|0x0|0x0|0x11|0x1|0x>
UDP/AutoRemote connected
GET data to URL: /sendmessage?message=Node;1;Bawah
closing connection
0
No API at all, so Skip!
(MaCaco)<OUT><|0x15|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB01><|0x23|0x17|0xAB01|0xAB02>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB02|0xAB01|0x5|0x0|0x0|0x57|0x18>
(MaCaco)<IN><|0x5|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x15|0x0|0x0|0x57|0x18|0

Multicast ON relay
(vNet)<IN>(0xAB01)<|0xC|0x17|0xFFFF|0xAB01|0x34|0x0|0x0|0x11|0x1|0x2>
(MaCaco)<IN><|0x34|0x0|0x0|0x11|0x1|0x>
UDP/AutoRemote connected
GET data to URL: /sendmessage?message=Node;0;Bawah
closing connection
0
No API at all, so Skip!
(MaCaco)<OUT><|0x15|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB01><|0x23|0x17|0xAB01|0xAB02>

juanpintom, 2015-08-18:
Good!

plinioseniore, 2015-08-18:
Hi Les,

have you loaded this new code on all your nodes? Please let me know test all your cases and let me know.

Dario.

plinioseniore, 2015-08-18:
I've updated the wiki page with some details
https://github.com/souliss/souliss/wiki/Gateway

lesjaw, 2015-08-18:
Yes I am uploading my test code on all my node... No more network flood...
It run good... I will do more test... If I found bug I will report asap..

lesjaw, 2015-08-19:
Dario, I got strange problem..

It seems, the peer node now having the hardtime to connect to SoulissApp, I
often got bad health and can not control the node, and when i check it i
can't ping it too..

at first everything is right, then I install SoulissApp on my second
Android, node got detected well.. but after couple of minute my second
android never got the value updated, the state of the typicals is wrong, in
my first android everything run as it suppose to.. the feedback from the
node is coming so the state of the typicals is the right value..

but i cant make it work in my second android, the health is good (255)
refreshing the node state by swiping the screen to the button take forever,
then i clear the DB and update DB node, all the node got detected well, but
still I got no feedback from the node, refreshing the node takes forever,
except for Node0, the feedback sometime good sometime dont..

i have a temp node it shows always 0 in my second android but in my first
android it show the right value and i can see it always updated..

and now this is the strange problem come, then I uninstall SoulissApp in my
second Android, install SoulissApp again, restart all node and now
everything works.. both of SoulissApp in my android works as it suppose to
be..

and I cant replicate the error, hmm .. I will test this until the rest of
the day.. will report again tonight..

plinioseniore, 2015-08-19:
SoulissApp has two network parameters defined at install time, ensure that
each SoulissApp installation has different ones (are in Network menu).

Let me know.

lesjaw, 2015-08-19:
yes .. i am checking that setting in both android.. btw it seems working
fine now.. both android got the same update time..

lesjaw, 2015-08-20:
yesterday I got some problem on connecting a new peer to the gateway, I
need to erase the eeprom of the gateway then start the gateway node again..
and then the peer can connect to the gateway..

Now today I have a new peer.. and the behaviour seem replicated.. my new
peer stuck on getting vNet address

here is the debug log..

Starting Souliss Node
Reading Configuration
Configuration Found!
AP mode started
192.168.4.1
Configuring Wifi
Connecting to AP.
.
.
.
.
.
.
.
.
.
Already connect to : Bolt
Admin Time Out
STA mode started
192.168.0.110
Souliss update IP Address
(vNet)<SETADDR><|0x6E>
(IP)<C0,A8,0,6E,>
(vNet)<SETADDR><|0x6E>
(IP)<C0,A8,0,6E,>
HTTP server started
0
Peer Mode
(vNet)<SETADDR><|0xAB00>
(MaCaco)<OUT><|0x29|0xBB|0x1C|0xF3|0x0|0x>
(vNet)<BRD><Media><|0x1><|0xB|0x17|0xFFFF|0x6E>
(vNet)<BRD><Media><|0x3><|0xB|0x17|0xFFFF|0xAB00>
(MaCaco)<OUT><|0x29|0xBB|0x1C|0xF3|0x0|0x>
(vNet)<BRD><Media><|0x1><|0xB|0x17|0xFFFF|0x6E>
(vNet)<BRD><Media><|0x3><|0xB|0x17|0xFFFF|0xAB00>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB02|0xAB01|0x5|0x0|0x0|0x57|0x18>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB02)<|0x23|0x17|0xAB01|0xAB02|0x15|0x0|0x0|0x57|0x18|0x1|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB07|0xAB01|0x5|0x0|0x0|0x57|0x18>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB07)<|0x23|0x17|0xAB01|0xAB07|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB03|0xAB01|0x5|0x0|0x0|0x57|0x18>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB03)<|0x23|0x17|0xAB01|0xAB03|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB04|0xAB01|0x5|0x0|0x0|0x57|0x18>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB04)<|0x23|0x17|0xAB01|0xAB04|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB06|0xAB01|0x5|0x0|0x0|0x57|0x18>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB06)<|0x23|0x17|0xAB01|0xAB06|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x1|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<OUTPATH><|0x3>
(MaCaco)<OUT><|0x29|0xBB|0x1C|0xF3|0x0|0x>
(vNet)<BRD><Media><|0x1><|0xB|0x17|0xFFFF|0x6E>
(vNet)<BRD><Media><|0x3><|0xB|0x17|0xFFFF|0xAB00>
(MaCaco)<OUT><|0x29|0xBB|0x1C|0xF3|0x0|0x>
(vNet)<BRD><Media><|0x1><|0xB|0x17|0xFFFF|0x6E>
(vNet)<BRD><Media><|0x3><|0xB|0x17|0xFFFF|0xAB00>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB02|0xAB01|0x5|0x0|0x0|0x57|0x18>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB07|0xAB01|0x5|0x0|0x0|0x57|0x18>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB07)<|0x23|0x17|0xAB01|0xAB07|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB03|0xAB01|0x5|0x0|0x0|0x57|0x18>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB03)<|0x23|0x17|0xAB01|0xAB03|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB04|0xAB01|0x5|0x0|0x0|0x57|0x18>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB04)<|0x23|0x17|0xAB01|0xAB04|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB06|0xAB01|0x5|0x0|0x0|0x57|0x18>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB06)<|0x23|0x17|0xAB01|0xAB06|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x1|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<OUTPATH><|0x3>
(MaCaco)<OUT><|0x29|0xBB|0x1C|0xF3|0x0|0x>
(vNet)<BRD><Media><|0x1><|0xB|0x17|0xFFFF|0x6E>
(vNet)<BRD><Media><|0x3><|0xB|0x17|0xFFFF|0xAB00>

i tried restarting the gateway or the new peer, the problem still persist,
if i delete the gateway eeprom I am sure that new peer can connect to the
gateway..

lesjaw, 2015-08-20:
and this is the gateway debug

Starting SerbaAuto Node
Reading Configuration
Configuration Found!
AP mode started
192.168.4.1
Configuring Wifi
Connecting to AP.
.
.
.
.
.
.
.
.
.
Already connect to : Bolt
Admin Time Out
STA mode started
192.168.0.102
Souliss update IP Address
(vNet)<SETADDR><|0x66>
(IP)<C0,A8,0,66,>
(vNet)<SETADDR><|0x66>
(IP)<C0,A8,0,66,>
HTTP server started
Gateway Mode
(vNet)<SETADDR><|0xAB01>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB02><|0xB|0x17|0xAB02|0xAB01>
(vNet)<IN>(0xAB02)<|0x23|0x17|0xAB01|0xAB02|0x15|0x0|0x0|0x57|0x18|0x1|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB07><|0xB|0x17|0xAB07|0xAB01>
(vNet)<IN>(0xAB07)<|0x23|0x17|0xAB01|0xAB07|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB03><|0xB|0x17|0xAB03|0xAB01>
(vNet)<IN>(0xAB03)<|0x23|0x17|0xAB01|0xAB03|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x5>
(vNet)<NHBOR><|0x5>
(vNet)<MEDIA><|0x5>
(vNet)<OUT><DADDR><|0x5><|0xB|0x17|0xD002|0x0>
(vNet)<OUT>-FAIL
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB04><|0xB|0x17|0xAB04|0xAB01>
(vNet)<IN>(0xAB04)<|0x23|0x17|0xAB01|0xAB04|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB06><|0xB|0x17|0xAB06|0xAB01>
(vNet)<IN>(0x64)<|0x23|0x17|0x66|0xD002|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x50|0xF4|0x4E|0x71|0x38|0xB|0x47|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<ROUTEUPDATE><|0xD000|0x64>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(vNet)<IN>(0xAB00)<|0xB|0x17|0xFFFF|0xAB00|0x29|0xBB|0x1C|0xF3|0x0>
(MaCaco)<IN><|0x29|0xBB|0x1C|0xF3|0x0|0x>

ctx: cont
sp: 3ffec510 end: 3ffec780 offset: 01b0

>>>stack>>>
3ffec6c0:  3ffec7e0 00000002 3ffeb394 402109f8
3ffec6d0:  3ffec7e0 3ffeb388 3ffeb394 402034cc
3ffec6e0:  3ffeb44c 3ffeb394 3ffeb3d8 3ffeafec
3ffec6f0:  3ffe800b 0000ab00 3ffeb388 4020a448
3ffec700:  4021b352 3ffea71c 3ffeb388 3ffea650
3ffec710:  00000001 3ffeafe6 3ffea684 4020a48c
3ffec720:  3ffeafec 3ffe800b 3ffe800d 40209688
3ffec730:  00000004 3ffea71c 3ffea684 4020af78
3ffec740:  00000000 00000000 00000016 40102171
3ffec750:  40201811 00000000 00000000 3ffec7ac
3ffec760:  3fffdc20 00000000 3ffec7a4 4020186e
3ffec770:  00000000 00000000 3ffeb760 40100378
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(1,0)

lesjaw, 2015-08-20:
ups my bad..

I add GetAsGateway(); now no more gateway crash..

plinioseniore, 2015-08-20:
Hi Les,

isn't clear for me which is the problem. If I look at the Peer log, I see a request for a new address that never gets a response from the Gateway, but in order to understand what's happening we need the log from the Gateway and Peer collected at same time.

Without I cannot understand what the Gateway does.

What about the crash? The GetAsGateway() just returns if the node is set as Gateway and isn't mandatory. There is no direct relation between this and a crash.

Dario.

plinioseniore, 2015-08-20:
Hi Les,

please print on the Serial what you get from GetAsGateway() because I suspect that the EEPROM values may overwrite the SetAsGateway and this result into a node that no longer act as a Gateway.

Let me know,
Dario.

lesjaw, 2015-08-21:
Hi Dario.. ok will report soon.. Right now I am running out power supply...
But as soon I got psu on my hands. I will test this again..

lesjaw, 2015-08-21:
yes Dario..GetAsGetway didn't fix my gateway crash..

I just another peer node and I saw the gateway crash again on vNet address
peer request

here is the gateway log

Reading Configuration
Configuration Found!
AP mode started
192.168.4.1
Configuring Wifi
Connecting to AP.
.
.
.
.
.
.
.
.
.
Already connect to : Bolt
Admin Time Out
STA mode started
192.168.0.102
Souliss update IP Address
(vNet)<SETADDR><|0x66>
(IP)<C0,A8,0,66,>
(vNet)<SETADDR><|0x66>
(IP)<C0,A8,0,66,>
HTTP server started
Gateway Mode
(vNet)<SETADDR><|0xAB01>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB02><|0xB|0x17|0xAB02|0xAB01>
(vNet)<IN>(0xAB02)<|0x23|0x17|0xAB01|0xAB02|0x15|0x0|0x0|0x57|0x18|0x1|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB07><|0xB|0x17|0xAB07|0xAB01>
(vNet)<IN>(0xAB07)<|0x23|0x17|0xAB01|0xAB07|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB03><|0xB|0x17|0xAB03|0xAB01>
(vNet)<IN>(0xAB03)<|0x23|0x17|0xAB01|0xAB03|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x5>
(vNet)<NHBOR><|0x5>
(vNet)<MEDIA><|0x5>
(vNet)<OUT><DADDR><|0x5><|0xB|0x17|0xD002|0x0>
(vNet)<OUT>-FAIL
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB04><|0xB|0x17|0xAB04|0xAB01>
(vNet)<IN>(0xAB04)<|0x23|0x17|0xAB01|0xAB04|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
sending NTP packet...
NTP no packet yet
(vNet)<IN>(0x6D)<|0xB|0x17|0x66|0x1496|0x25|0x0|0x0|0x0|0x4>
(MaCaco)<IN><|0x25|0x0|0x0|0x0|0x4|0x>
(MaCaco)<OUT><|0x35|0x0|0x0|0x0|0x4|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x1496><|0xF|0x17|0x1496|0x66>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB06><|0xB|0x17|0xAB06|0xAB01>
(vNet)<IN>(0xAB06)<|0x23|0x17|0xAB01|0xAB06|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB08><|0xB|0x17|0xAB08|0xAB01>
(vNet)<IN>(0xAB08)<|0x23|0x17|0xAB01|0xAB08|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x1|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(vNet)<IN>(0xAB07)<|0x23|0x17|0xAB01|0xAB07|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x1|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(vNet)<IN>(0xAB07)<|0x23|0x17|0xAB01|0xAB07|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(vNet)<IN>(0x6D)<|0xB|0x17|0x66|0x1496|0x21|0x0|0x0|0x0|0x4>
(MaCaco)<IN><|0x21|0x0|0x0|0x0|0x4|0x>
(MaCaco)<OUT><|0x31|0x0|0x0|0x0|0x18|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x1496><|0x23|0x17|0x1496|0x66>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB06><|0xB|0x17|0xAB06|0xAB01>
(vNet)<IN>(0xAB06)<|0x23|0x17|0xAB01|0xAB06|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x31|0x0|0x0|0x6|0x18|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x1496><|0x23|0x17|0x1496|0x66>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB08><|0xB|0x17|0xAB08|0xAB01>
(vNet)<IN>(0xAB08)<|0x23|0x17|0xAB01|0xAB08|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x1|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x31|0x0|0x0|0x7|0x18|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x1496><|0x23|0x17|0x1496|0x66>
(vNet)<IN>(0xAB00)<|0xB|0x17|0xFFFF|0xAB00|0x29|0x14|0x1C|0xF3|0x0>
(MaCaco)<IN><|0x29|0x14|0x1C|0xF3|0x0|0x>

ctx: cont
sp: 3ffec510 end: 3ffec790 offset: 01b0

>>>stack>>>
3ffec6c0:  3ffec7f0 00000002 3ffeb3a4 40210a3c
3ffec6d0:  3ffec7f0 3ffeb398 3ffeb3a4 402034cc
3ffec6e0:  3ffeb45c 3ffeb3a4 3ffeb3e8 3ffeaffc
3ffec6f0:  3ffe800b 0000ab00 3ffeb398 4020a448
3ffec700:  3ffec7f0 3ffea72c 3ffeb398 3ffea708
3ffec710:  3ffeaff0 3ffeaff6 3ffea694 4020a48c
3ffec720:  3ffeaffc 3ffe800b 3ffe800d 3ffea708
3ffec730:  3ffeaff0 3ffea72c 3ffea694 4020afbc
3ffec740:  3ffe9d00 6800a8c0 40203104 40202f84
3ffec750:  00000000 00000000 00000016 40102171
3ffec760:  40201811 00000000 00000000 3ffec7bc
3ffec770:  3fffdc20 00000000 3ffec7b4 4020186e
3ffec780:  00000000 00000000 3ffeb770 40100378
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x4010f000, len 1264, room 16
tail 0
chksum 0x42
csum 0x42
~ld
Starting SerbaAuto Node
Reading Configuration
Configuration Found!
AP mode started
192.168.4.1
Configuring Wifi
Connecting to AP

lesjaw, 2015-08-21:
and this is the peer log..

Reading Configuration
Configuration Found!
AP mode started
192.168.4.1
Configuring Wifi
Connecting to AP.
.
.
.
.
.
.
.
.
.
Already connect to : Bolt
Admin Time Out
STA mode started
192.168.0.110
Souliss update IP Address
(vNet)<SETADDR><|0x6E>
(IP)<C0,A8,0,6E,>
(vNet)<SETADDR><|0x6E>
(IP)<C0,A8,0,6E,>
HTTP server started
0
Peer Mode
(vNet)<SETADDR><|0xAB00>
(MaCaco)<OUT><|0x29|0x14|0x1C|0xF3|0x0|0x>
(vNet)<BRD><Media><|0x1><|0xB|0x17|0xFFFF|0x6E>
(vNet)<BRD><Media><|0x3><|0xB|0x17|0xFFFF|0xAB00>

*it seems the gateway crash is here..*

*then i continue it ..*

(vNet)<IN>(0xAB03)<|0x23|0x17|0xAB01|0xD002|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x50|0x50|0x90|0x4F|0x71|0x38|0xB|0x47|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<ROUTEUPDATE><|0xD000|0xAB03>
(vNet)<OUTPATH><|0x3>
(MaCaco)<OUT><|0x29|0x14|0x1C|0xF3|0x0|0x>
(vNet)<BRD><Media><|0x1><|0xB|0x17|0xFFFF|0x6E>
(vNet)<BRD><Media><|0x3><|0xB|0x17|0xFFFF|0xAB00>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB02|0xAB01|0x5|0x0|0x0|0x57|0x18>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB02)<|0x23|0x17|0xAB01|0xAB02|0x15|0x0|0x0|0x57|0x18|0x1|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB07|0xAB01|0x5|0x0|0x0|0x57|0x18>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB07)<|0x23|0x17|0xAB01|0xAB07|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB03|0xAB01|0x5|0x0|0x0|0x57|0x18>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB03)<|0x23|0x17|0xAB01|0xAB03|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB04|0xAB01|0x5|0x0|0x0|0x57|0x18>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB04)<|0x23|0x17|0xAB01|0xAB04|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<OUTPATH><|0x3>
(MaCaco)<OUT><|0x29|0x14|0x1C|0xF3|0x0|0x>
(vNet)<BRD><Media><|0x1><|0xB|0x17|0xFFFF|0x6E>
(vNet)<BRD><Media><|0x3><|0xB|0x17|0xFFFF|0xAB00>
(vNet)<IN>(0xAB07)<|0x23|0x17|0xAB01|0xAB07|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x1|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB07)<|0x23|0x17|0xAB01|0xAB07|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<OUTPATH><|0x3>
(MaCaco)<OUT><|0x29|0x14|0x1C|0xF3|0x0|0x>
(vNet)<BRD><Media><|0x1><|0xB|0x17|0xFFFF|0x6E>
(vNet)<BRD><Media><|0x3><|0xB|0x17|0xFFFF|0xAB00>
(vNet)<IN>(0xAB03)<|0x23|0x17|0xAB01|0xD002|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x50|0x50|0x90|0x4F|0xB4|0x37|0x18|0x46|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<OUTPATH><|0x3>
(MaCaco)<OUT><|0x29|0x14|0x1C|0xF3|0x0|0x>
(vNet)<BRD><Media><|0x1><|0xB|0x17|0xFFFF|0x6E>
(vNet)<BRD><Media><|0x3><|0xB|0x17|0xFFFF|0xAB00>
(vNet)<IN>(0xAB03)<|0x23|0x17|0xAB01|0xD002|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x50|0x50|0x90|0x4F|0x9|0x39|0x0|0x48|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB03)<|0x23|0x17|0xAB01|0xD002|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x50|0x50|0x90|0x4F|0x71|0x38|0xB|0x47|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<OUTPATH><|0x3>
(MaCaco)<OUT><|0x29|0x14|0x1C|0xF3|0x0|0x>
(vNet)<BRD><Media><|0x1><|0xB|0x17|0xFFFF|0x6E>
(vNet)<BRD><Media><|0x3><|0xB|0x17|0xFFFF|0xAB00>
(MaCaco)<OUT><|0x29|0x14|0x1C|0xF3|0x0|0x>
(vNet)<BRD><Media><|0x1><|0xB|0x17|0xFFFF|0x6E>
(vNet)<BRD><Media><|0x3><|0xB|0x17|0xFFFF|0xAB00>
(vNet)<IN>(0xAB03)<|0x23|0x17|0xAB01|0xD002|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x50|0x50|0x90|0x4F|0xB4|0x37|0x18|0x46|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<OUTPATH><|0x3>
(vNet)<IN>(0xAB02)<|0xB|0x17|0xFFFF|0xAB02|0x2B|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x2B|0x0|0x0|0x0|0x0|0x>
(MaCaco)<OUT><|0x29|0x14|0x1C|0xF3|0x0|0x>
(vNet)<BRD><Media><|0x1><|0xB|0x17|0xFFFF|0x6E>
(vNet)<BRD><Media><|0x3><|0xB|0x17|0xFFFF|0xAB00>

juanpintom, 2015-08-21:
I suggest add this at loop() with NTP enabled, will be more easy track
errors, do you think?

 if (Refresh)
{
Refresh = false;
   Serial.printf("FreeMem:%d %d:%d:%d %d.%d.%d \n",ESP.getFreeHeap() ,
DateTime.hour,DateTime.minute, DateTime.second, DateTime.year,
DateTime.month, DateTime.day);
         }

Regards

lesjaw, 2015-08-21:
Currently my esp network have no Internet connection Juan.. Can't update
ntp yet..

So do you think it is because free men,?

juanpintom, 2015-08-21:
Isn't a memory problem, but with NTP time Dario can know when happens what
:P

Regards

plinioseniore, 2015-08-21:
Hi Les,

you are probably out of memory. You should try a standard sketch with
dynamic addressing, if it crash we should look at the Souliss code.

Does your node restart automatically?

Please post on Arduino/esp8266 github the message after the crash, they may
help on root cause.

Dario.

From Mobile.

lesjaw, 2015-08-21:
Oh I see.. hheheehehe

here is the log :

Refreshing...
FreeMem:19552 6:1:59 1970.1.1
(vNet)<IN>(0xAB08)<|0xB|0x17|0xFFFF|0xAB08|0x2C|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x2C|0x0|0x0|0x0|0x0|0x>
(vNet)<IN>(0xAB04)<|0xB|0x17|0xFFFF|0xAB04|0x2C|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x2C|0x0|0x0|0x0|0x0|0x>
(vNet)<IN>(0xAB06)<|0xB|0x17|0xFFFF|0xAB06|0x2C|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x2C|0x0|0x0|0x0|0x0|0x>
Refreshing...
FreeMem:19552 6:2:0 1970.1.1
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB02><|0xB|0x17|0xAB02|0xAB01>
(vNet)<IN>(0xAB07)<|0xB|0x17|0xFFFF|0xAB07|0x2C|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x2C|0x0|0x0|0x0|0x0|0x>
(vNet)<IN>(0xAB08)<|0xB|0x17|0xFFFF|0xAB08|0x2C|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x2C|0x0|0x0|0x0|0x0|0x>
(vNet)<IN>(0xAB02)<|0x23|0x17|0xAB01|0xAB02|0x15|0x0|0x0|0x57|0x18|0x1|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x31|0x0|0x0|0x1|0x18|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x5EC8><|0x23|0x17|0x5EC8|0x66>
Refreshing...
FreeMem:19552 6:2:1 1970.1.1
(vNet)<IN>(0xAB04)<|0xB|0x17|0xFFFF|0xAB04|0x2C|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x2C|0x0|0x0|0x0|0x0|0x>
(vNet)<IN>(0xAB06)<|0xB|0x17|0xFFFF|0xAB06|0x2C|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x2C|0x0|0x0|0x0|0x0|0x>
(vNet)<IN>(0xAB03)<|0xB|0x17|0xFFFF|0xAB03|0x2B|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x2B|0x0|0x0|0x0|0x0|0x>
Refreshing...
FreeMem:19552 6:2:2 1970.1.1
(vNet)<IN>(0xAB07)<|0xB|0x17|0xFFFF|0xAB07|0x2C|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x2C|0x0|0x0|0x0|0x0|0x>
(vNet)<IN>(0xAB08)<|0xB|0x17|0xFFFF|0xAB08|0x2C|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x2C|0x0|0x0|0x0|0x0|0x>
(vNet)<IN>(0xAB00)<|0xB|0x17|0xFFFF|0xAB00|0x29|0x14|0x1C|0xF3|0x0>
(MaCaco)<IN><|0x29|0x14|0x1C|0xF3|0x0|0x>

ctx: cont
sp: 3ffec530 end: 3ffec7c0 offset: 01b0

>>>stack>>>
3ffec6e0:  3ffec820 00000002 3ffeb3d4 40210a80
3ffec6f0:  3ffec820 3ffeb3c8 3ffeb3d4 402034cc
3ffec700:  3ffeb48c 3ffeb3d4 3ffeb418 3ffeb02c
3ffec710:  3ffe800b 0000ab00 3ffeb3c8 4020a448
3ffec720:  3ffe9c44 3ffea6c4 3ffeb3c8 3ffea738
3ffec730:  3ffeb020 3ffea6c4 3ffeb026 4020a48c
3ffec740:  3ffeb02c 3ffe800b 3ffec820 40210b38
3ffec750:  3ffeb020 3ffea6c4 00000001 4020affc
3ffec760:  000007b2 00000001 00000001 4020aa2e
3ffec770:  3ffe9d30 6800a8c0 40203104 40202f84
3ffec780:  00000000 00000000 00000016 40102171
3ffec790:  40201811 00000000 00000000 3ffec7ec
3ffec7a0:  3fffdc20 00000000 3ffec7e4 4020186e
3ffec7b0:  00000000 00000000 3ffeb7a0 40100378
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(1,0)

lesjaw, 2015-08-21:
Is this normall for peer to send its address as 0xAB00?

HTTP server started
0
Peer Mode
(vNet)<SETADDR><|0xAB00>
(MaCaco)<OUT><|0x29|0x14|0x1C|0xF3|0x0|0x>
(vNet)<BRD><Media><|0x1><|0xB|0x17|0xFFFF|0x6E>
(vNet)<BRD><Media><|0x3><|0xB|0x17|0xFFFF|0xAB00>

as I can see the data from the peer is coming in the gateway, but before
the gateway has the chance to proceed the vnet address it got crashed :

Refreshing...
FreeMem:19680 6:14:37 1970.1.1
Refreshing...
FreeMem:19680 6:14:38 1970.1.1
Refreshing...
FreeMem:19680 6:14:39 1970.1.1
Refreshing...
FreeMem:19680 6:14:40 1970.1.1
(vNet)<IN>(0xAB00)<|0xB|0x17|0xFFFF|0xAB00|0x29|0x14|0x1C|0xF3|0x0>
(MaCaco)<IN><|0x29|0x14|0x1C|0xF3|0x0|0x>

ctx: cont
sp: 3ffec530 end: 3ffec7c0 offset: 01b0

>>>stack>>>
3ffec6e0:  3ffec820 00000002 3ffeb3d4 40210a80
3ffec6f0:  3ffec820 3ffeb3c8 3ffeb3d4 402034cc
3ffec700:  3ffeb48c 3ffeb3d4 3ffeb418 3ffeb02c
3ffec710:  3ffe800b 0000ab00 3ffeb3c8 4020a448
3ffec720:  3ffe9c44 3ffea6c4 3ffeb3c8 3ffea738
3ffec730:  3ffeb020 3ffea6c4 3ffeb026 4020a48c
3ffec740:  3ffeb02c 3ffe800b 3ffec820 40210b38
3ffec750:  3ffeb020 3ffea6c4 00000001 4020affc
3ffec760:  000007b2 00000001 00000001 4020aa2e
3ffec770:  3ffe9d30 6600a8c0 40203104 40202f84
3ffec780:  00000000 00000000 00000016 40102171
3ffec790:  40201811 00000000 00000000 3ffec7ec
3ffec7a0:  3fffdc20 00000000 3ffec7e4 4020186e
3ffec7b0:  00000000 00000000 3ffeb7a0 40100378
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x4010f000, len 1264, room 16
tail 0
chksum 0x42
csum 0x42
~ld
Starting Souliss Node
Reading Configuration

lesjaw, 2015-08-21:
Dario, & Juan..

Ok this is still guessing, so I guess the problem is in the gateway logic
on providing the vNet address..

so this is the case :

I have 8 nodes, 1 gateway and 7 peer..

at first boot of setup... if I am doing a refresh setup, all node run
correctlly.. but after it runs a couple of days then I add another peer
node the problem begin, the gateaway crash on vNet request address...
(happens 2 times now)

*Some explanation (making example with 3 Nodes):*
Node 0(gateway) has T11 and Node 1(peer) has T11 too, by seeing the vNet
debug, Node 0 vNet address is 0xAB01 (which is normal and it should be
always 0xAB01 as gateway right?) and Node 1 is 0xAB02...

Sometime I saw the *same* vNet address on another peer node, let said I add
another peer node, and it got vNet address 0xAB02 too, this is wierd and in
SoulissApp now Node 1 has 2 typicals, if the second peer use different slot
it will act normal.. so basically it likes 1 device (node) with 2 typicals,
infact it is 2 devices and with each one typicals inside.

I have that such case randomly, if the other device use the same slot then
it will conflict with the first peer node, to fix this I must erase eeprom
gateway and the second peer node (the third node), and in the next boot it
will get the right address, sometimes the third node (second peer) got the
vNet address as 0xAB05 instead 0xAB03, since 0xAB01 as gateway and the
second node got vNet address as 0xAB02 so the next vNet address of the
third node should be 0xAB03 right?

And right now I have that case, what I am doing is I am erasing my last
peer node (which is causing gateway crash) and set it as gateway.. and my
previous gateway (the one that got crash) set as peer, now I got 2 Nodes
with vNet address as 0xAB01 and 0xAB02, so far this is good...

then I try add another peer, this peer was the Node 1 with vNet address
0xAB02, remember on the previous setup this node was already connect to the
gateway and have vNet address as 0xAB02.. and this is the problem begin..
my new Gateway never got this vNet request address, I guess because it
already has vNet address inside the eeprom which is 0xAB02, and this
gateway already setup the 0xAB02 for another node..

Since this both node has the same typicals and the slot, so the STATUS on
SoulissApp is normal, right now if I press ON or OFF (T11) both of the
devices will act the same..so now Node 1 in SoulissAPp is controling 2
Nodes (devices). but if it has different typicals then the status if the
node in SoulissApp will UNKNOWN.

*Behaviour of Gateway crash:*
Lets see this behaviour of my gateway crash, it seems only happen if I add
new/fresh esp from the box, if it ever connect to the gateway then it will
be just fine in the request vNet address, but the most confuse thing
happens to me, if I erase the gateway eeprom then this new/fresh esp will
got recognize by the gateway, and it will get the new vNet address (no
double vNet address).

thats all I can report for now.. hope we can find a solution for this..

thanks

lesjaw, 2015-08-21:
Here is the gateway log, so I erase my gateway and flash the sketch again,
it is T42..

Already connect to : Bolt
Admin Time Out
STA mode started
192.168.0.101
Souliss update IP Address
(vNet)<SETADDR><|0x65>
(IP)<C0,A8,0,65,>
(vNet)<SETADDR><|0x65>
(IP)<C0,A8,0,65,>
HTTP server started
Gateway Mode
(vNet)<SETADDR><|0xAB01>
192.168.0.104
(vNet)<IN>(0x68)<|0xB|0x17|0x65|0x5EC8|0x27|0x0|0x0|0x1|0x1>
(MaCaco)<IN><|0x27|0x0|0x0|0x1|0x1|0x>
(vNet)<IN>(0xAB02)<|0xB|0x17|0xFFFF|0xAB02|0x2C|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x2C|0x0|0x0|0x0|0x0|0x>
(MaCaco)<ADDRS><|0x65|0xAB02|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB02><|0xB|0x17|0xAB02|0xAB01>
(vNet)<IN>(0xAB02)<|0x23|0x17|0xAB01|0xAB02|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>

as you can see, after the gateway running, then I turn on my second node
(first Peer Node) which has ever connected to the gateway, and it has
0xAB02 as its vNet address..

Next, I am going to clear eeprom of my second peer node, it has T11
inside..I will report the log..

lesjaw, 2015-08-21:
and this us the gateway log,

(vNet)<IN>(0xAB00)<|0xB|0x17|0xFFFF|0xAB00|0x29|0x33|0x1D|0xF3|0x0>
(MaCaco)<IN><|0x29|0x33|0x1D|0xF3|0x0|0x>
(MaCaco)<OUT><|0x39|0x33|0x1D|0x3|0x2|0x>
(vNet)<BRD><Media><|0x1><|0xD|0x17|0xFFFF|0x65>
(vNet)<BRD><Media><|0x3><|0xD|0x17|0xFFFF|0xAB01>
(vNet)<IN>(0xAB03)<|0xB|0x17|0xFFFF|0xAB03|0x2C|0x33|0x1D|0x0|0x0>
(MaCaco)<IN><|0x2C|0x33|0x1D|0x0|0x0|0x>
(MaCaco)<ADDRS><|0x65|0xAB02|0xAB03|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB02><|0xB|0x17|0xAB02|0xAB01>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB03><|0xB|0x17|0xAB03|0xAB01>
(vNet)<IN>(0xAB02)<|0x23|0x17|0xAB01|0xAB02|0x15|0x0|0x0|0x57|0x18|0x1|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x31|0x0|0x0|0x1|0x18|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x5EC8><|0x23|0x17|0x5EC8|0x65>
(vNet)<IN>(0xAB03)<|0x23|0x17|0xAB01|0xAB03|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x31|0x0|0x0|0x2|0x18|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x5EC8><|0x23|0x17|0x5EC8|0x65>

the vNet request is fine right? My third node (second peer) got the right
vNet address 0xAB03..

this is only happen if I erase the eeprom of the gateway and erase eeprom
of the peer node (the one that before make gateway crash), right now I have
another 5 peer nodes OFF, which one of them already has 0xAB03, if i turn
it ON I am sure it will not request vNet address, once it ON it will
conflict with current 0xAB03 since it has differen typical but use the same
slot.

but anyway, I am still can't find why the gateway crash if I add new fresh
esp from the box, by erasing the eeprom it should have the same fresh
behaviour right? so logically by adding this third node (second peer) my
gateway should crash, but it did not..

Now I am going to try erase all my 5 current nodes, see if any of them make
my gateway crash..

lesjaw, 2015-08-22:
Hmm.. I was erasing my peer node, it was has 0xAB06..  now after it connect
to gateway it got 0xAB04.. it seems works fine.. no gateway crash..

I think I know why sometime the vNet address jumped from 0xAB03 to 0xAB07
or anyrhing else, it was just because the peer already have vNet address..
so the gateway vNet DHCP logic still right..

(MaCaco)<OUT><|0x39|0xF|0x1E|0x3|0x2|0x>
(vNet)<BRD><Media><|0x1><|0xD|0x17|0xFFFF|0x65>
(vNet)<BRD><Media><|0x3><|0xD|0x17|0xFFFF|0xAB01>
(vNet)<IN>(0xAB04)<|0xB|0x17|0xFFFF|0xAB04|0x2C|0xF|0x1E|0x0|0x0>
(MaCaco)<IN><|0x2C|0xF|0x1E|0x0|0x0|0x>
(MaCaco)<ADDRS><|0x65|0xAB02|0xAB03|0xAB04|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB02><|0xB|0x17|0xAB02|0xAB01>
(vNet)<IN>(0xAB02)<|0x23|0x17|0xAB01|0xAB02|0x15|0x0|0x0|0x57|0x18|0x1|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x31|0x0|0x0|0x1|0x18|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x5EC8><|0x23|0x17|0x5EC8|0x65>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB03><|0xB|0x17|0xAB03|0xAB01>
(vNet)<IN>(0xAB03)<|0x23|0x17|0xAB01|0xAB03|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x31|0x0|0x0|0x2|0x18|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x5EC8><|0x23|0x17|0x5EC8|0x65>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB04><|0xB|0x17|0xAB04|0xAB01>
(vNet)<IN>(0xAB04)<|0x23|0x17|0xAB01|0xAB04|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x31|0x0|0x0|0x3|0x18|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x5EC8><|0x23|0x17|0x5EC8|0x65>

I still cant replicate the gateway crash

lesjaw, 2015-08-22:
Ok..I found the problem and I can replicate the gateway crash..

So, Yes, my new esp from out of the box after I upload the sketch and in
the process of GetIPaddress make my gateway crash, this will not happen if
the esp was ever act like a node and once connected to the gateway, so it
has the vNet address inside.. to fix this I must erase my eeprom gateway,
then re-add all the peer, and add the last peer node.. until here
everything works normal, but if I add another peer from out of the box then
the gateway crash is begin, to fix this I must repeat the process above
(erase eeprom gateway)..

But this fix is not solution, how if I add another peer node, I must do the
whole process again.. so I need to know what causing this, and I must
enable replicate the error with my old peer, why it is only happen with the
esp from out of the box (fresh one)? why if i delete/erase eeprom my old
peer didn't have the same behavaiour?

So I reflash my old peer with the AT 9.5.2 firmware, then I can replicate
the crash, so deleting the eeprom of m old esp just not enough, to make it
like a fresh esp I must flash the espressif frimware..

And in the process of that I found another cause of the gateway crash,
logically the crash happen with new fresh esp, so  I should enable
replicate this with just two node (gateway and peer), so I flash this both
esp, and re-upload the sketch, deleting SoulissApp..

and Viola, no more crash, now the fresh esp didnt make the gateway crash, I
add another (third) fresh esp and try to make it to connect to gateway, and
Bam! still no crash, the gateway succeed give vNet address to the new fresh
esp..

So why does the gateway crash? it happen when I already have a lot of node
(6 or 8 nodes) and the vNet address is not in sequence number, it doesn't
happen if I use the clean gateway and clean peer (never connect to any
gateway)...

Basically we can't just turn on the peer as we wish, the gateway will
stable if the address request in consecutive manner (0xAB01 - 0xAB02 -
0xAB03), so the vNet address can't jump from 0xAB03 to suddenly
0xAB07...let said we have this node in this order :
Node 0 - 0xAB01
Node 1 - 0xAB02
Node 2 - 0xAB03
and so on...

with that structure of node the gateway will stable enough, adding a new
peer will work flawless no matter its a new or the old esp.. but if we have
node structure like this below :
Node 0 - 0xAB01
Node 1 - 0xAB02
Node 2 - 0xAB05
Node 3 - 0xAB04

then adding a new fresh esp will make the gateway crash.. but it will not
crash if the peer already once connect to gateway and has a vNet address,
it just a matter of the position of the node we will see, if our current
node structure or gateway doesn't have the peer address it will create a
new node, but if the gateway have the same address then it wll add a new
typicals inside the node.

So to make sure the gateway stable enough I must make sure the peer has no
vNet address inside..and if I can when I "Get Souliss DB" I must do this in
sequence manner..

Does it make sense Dario?

lesjaw, 2015-08-22:
Dario..

in the second structure..

Node 0 - 0xAB01
Node 1 - 0xAB02
Node 2 - 0xAB05
Node 3 - 0xAB04

assume this a fresh gateway :
1. I add a fresh esp from out of the box, no vNet address inside, then
gateway will give its vNet address as 0xAB02
2. I turn ON another peer, which was ever connect to *another* gateway, and
it has 0xAB05 vNet address, so the gateway will also give the vNet address
as it is, no new vNet address.
3. I turn On another peer, which also once connect to another gateway, and
it has 0xAB04 vNet address

so in the next peer node add, if it is a peer which has vNet address inside
then the gateway is OK, but if it is a new one (fresh not just by clearing
eeprom) and doesn't have vNet address then the gateway will run the logic
to provide the new vNet address for that peer, this is where the gateway
crash..

But that gateway vNet logic will not make the gateway crash if the gateway
has consecutive vNet address number..

thats the summary of my test..

and this lead to another problem, since I sale my souliss board then after
I do the QC (quality control), I am really have to make sure it doesn't
contain any data in the eeprom..

lesjaw, 2015-08-22:
here is the log when I add a new peer (I only clear its eeprom) and it got
new vNet address, and it is 0xAB04

(vNet)<IN>(0xAB00)<|0xB|0x17|0xFFFF|0xAB00|0x29|0x72|0x1C|0xF3|0x0>
(MaCaco)<IN><|0x29|0x72|0x1C|0xF3|0x0|0x>
(MaCaco)<OUT><|0x39|0x72|0x1C|0x3|0x2|0x>
(vNet)<BRD><Media><|0x1><|0xD|0x17|0xFFFF|0x66>
(vNet)<BRD><Media><|0x3><|0xD|0x17|0xFFFF|0xAB01>
(vNet)<IN>(0xAB04)<|0xB|0x17|0xFFFF|0xAB04|0x2C|0x72|0x1C|0x0|0x0>
(MaCaco)<IN><|0x2C|0x72|0x1C|0x0|0x0|0x>
(MaCaco)<ADDRS><|0x66|0xAB02|0xAB03|0xAB04|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB02><|0xB|0x17|0xAB02|0xAB01>
(vNet)<IN>(0xAB02)<|0x23|0x17|0xAB01|0xAB02|0x15|0x0|0x0|0x57|0x18|0x0|0x1|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB03><|0xB|0x17|0xAB03|0xAB01>
(vNet)<IN>(0xAB03)<|0x23|0x17|0xAB01|0xAB03|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB04><|0xB|0x17|0xAB04|0xAB01>
(vNet)<IN>(0xAB04)<|0x23|0x17|0xAB01|0xAB04|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(vNet)<IN>(0x6F)<|0xB|0x17|0x66|0x1496|0x21|0x0|0x0|0x0|0x4>
(MaCaco)<IN><|0x21|0x0|0x0|0x0|0x4|0x>
(MaCaco)<OUT><|0x31|0x0|0x0|0x0|0x18|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x1496><|0x23|0x17|0x1496|0x66>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB02><|0xB|0x17|0xAB02|0xAB01>
(vNet)<IN>(0xAB02)<|0x23|0x17|0xAB01|0xAB02|0x15|0x0|0x0|0x57|0x18|0x0|0x1|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x31|0x0|0x0|0x1|0x18|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x1496><|0x23|0x17|0x1496|0x66>

lesjaw, 2015-08-22:
And this is where I think where eventually this behaviour will become a
problem..

let said we already have :

Node 0 - 0xAB01
Node 1 - 0xAB02
Node 2 - 0xAB03
Node 3 - 0xAB04

then suddenly the gateway down.. now we have vNet address inside of that
peer as :
Node 1 - 0xAB02
Node 2 - 0xAB03
Node 3 - 0xAB04

then I pich 0xAB02 as the gateway, now the structure will be
Node 1 - 0xAB01
Node 2 - 0xAB03
Node 3 - 0xAB04

and we miss 0xAB02, when I add a new peer I am sure this will make the
gateway crash..

lesjaw, 2015-08-22:
I am sorry I mean the structure will be :
Node 0 - 0xAB01
Node 1 - 0xAB03
Node 2 - 0xAB04

plinioseniore, 2015-08-22:
What's running as sketch on these nodes? Could you please try with a
standard sketch with dynamic addressing?

Please report the crash information in Arduino/esp8266 github issues, they
may probably help us to understand the crash reason.

Dario.

From Mobile.

plinioseniore, 2015-08-22:
Hi Les,

just got now the opportunity to read your comments. Let me say that for next time, you should use a less confusing approach, because it became very hard to follow your test.

So every time please describe the intial condition and then proceed by step, 1) 2) and so on. You should write as less as possible giving at same time all details, this will make easier to avoid misunderstandings.

Back to the problem, you have to consider that you are running a code by your own that is conflicting with the original scope. So actually there is no provision to change a node from gateway to peer at runtime without erase the EEPROM and restart the process of adding nodes.
You are pointing to a problem, so we should be sure that moving to one configuration to another will means to erase the EEPROM. I will take note for.

Regarding the crash, there is no real reason for a crash as far as I can understand. The node may give a wrong address but I cannot give a relation with a crash.

So, my suggestion is to stop for a while and wait that I merge the code from Juan. That one will be the base code to be used for further testing (without any other adding), so that we are sure about what we are using as library and sketches.

I will close this and open two new issues for those problems.

closed 2015-08-22 by plinioseniore

plinioseniore, 2015-08-22:
Hi Les, look at #161 and continue the discussion there.