souliss/souliss #161: Check addressing order in case of Dynamic Addressing

plinioseniore, 2015-08-22:
As noted in #156 by Les, there may be cases when the Gateway node provide a wrong address. Has been seen that in the Gateway nodes didn't get an incremental ordering and this may lead to a wrong assignment for new addresses.

plinioseniore, 2015-08-22:
The crash referred in #156 doesn't look a reset by a watchdog, but in any case I've found a possible endless loop.

In commit 72ce9b27 has a fix, this can be tested against the crashes identified before by Les.

lesjaw, 2015-08-22:
Yes... Ok I will test this asap.. 
To replicate my gateway crash is :

1. Create souliss network with only 4 nodes, 
2. Set one node to run as gateway
3. Add peer node one by one
Until here everything should work correctly, and we should have this vNet address :
0xAB01
0xAB02
0xAB03
0xAB04
4. Take gateway down
5. Set one of the peer node as gateway
Will report the test asap..

plinioseniore, 2015-08-22:
Hi Les,

if you remove the Gateway and configure a previous Peer as new Gateway, you have to erase the EEPROM, otherwise your Gateway will not get 0xAB01 as address.

So, in your code perform a full reset while selecting the node as Gateway or use an external option to force an EEPROM reset.

Dario.

plinioseniore, 2015-08-24:
Hi Les, 

any news?

Dario.

lesjaw, 2015-08-24:
Sorry dario i heaven"t test this yet .. will do it tonight..

juanpintom, 2015-08-24:
Dario, take care about how the EEPROM works on ESP, you need to call EEPROM.commit(); after a write to save values, maybe this causes problems.

Regards.

lesjaw, 2015-08-25:
Dario .. it seems Ok now.. I still need to clean the eeprom after mode
change (gateway/peer)...

I will report again if I found some strange behaviour...

closed 2015-08-29 by plinioseniore