souliss/souliss #168: Add strict mode

shineangelic, 2015-09-04:
As a "front door security" add strict mode to the framework. If set, souliss would accept datagram ONLY from specified couples nodeIndex/UserIndex, ignoring the other requests.

juanpintom, 2015-09-04:
Good!, Security is one of the most asked questions from the users.

Ty for your work :P

Regards

plinioseniore, 2015-09-04:
As feasibility note, the incoming request from user interface are tagged at vNet driver level with something like:

```C
uint16_t umrec = ((*(U16*)&data_pnt[5]) & 0xFF00);
if((umrec != 0x0000) && (umrec <= VNET_ADDR_H_M1))
	UserMode_Record((*(U16*)&data_pnt[5]), dataframe.ip, (uint8_t *)(&dataframe.port));	
```

Each driver has its own, so to avoid several changed this lock-down can be implemented in UserMode (that is common for all transceivers) using a blacklist approach based on IP address. So an IP address that tries a connection using a not recorded vNet address will be blacklisted till node reboot.

To include this without effort on the sketches, we could allow new user interface in the first minutes form the boot, once over no additional user interfaces will be allowed till reboot. A record in the EEPROM may help to keep configuration persistent across reboots.

shineangelic, 2015-09-04:
'new user interface in the first minutes form the boot' is bad when someone has to smoke something before proceeding. I mean that every kind of security setting should be permanent, working 24x7 and protecting you from 'most basic' attacks without time constraints, even if very weak.

Adding parameters to soulissApp is fast and straightforward; IMHO we should not consider real security scenarios here, it's just a matter of front door stuff, you mildly block someone who knows the protocol but ignore your userIndex.

plinioseniore, 2015-09-05:
If you want, you can lower that minute to zero seconds, and get the behavior that you want. Actually it works either in run-time and compile-time.

https://github.com/souliss/souliss/wiki/Security%20and%20Remote%20Access

plinioseniore, 2015-09-05:
In commit 9cbae03b is now available the Lockdown, tested.

closed 2015-09-05 by plinioseniore

juanpintom, 2015-09-07:
> In commit 9cbae03
> is now available the Lockdown, tested.

Very Good Work!, I'll include on my network soon as possible.

Regards