souliss/souliss #175: Deep test on T31

plinioseniore, 2015-09-18:
As per @fazioa test, there are some odd behavior in T31. Actually he is matching the commit 0d1f578353 with the actual HEAD of friariello.

That commit has no related modification to T31, so is something related to previous commit, [this](https://github.com/souliss/souliss/compare/0d1f5783539183169343c15a18d898101ef22d8b...friariello) is a direct comparison between these commits.

Other resources,
[T31.cpp file at 01df57](https://github.com/souliss/souliss/blame/0d1f5783539183169343c15a18d898101ef22d8b/base/T3n.cpp)
[T31.cpp at HEAD](https://github.com/souliss/souliss/blame/friariello/base/T3n.cpp)

Lets discuss here about

fazioa, 2015-09-18:
Line 122
Revert to
if(C8TO16(memory_map + MaCaco_IN_s + slot) != Souliss_T3n_RstCmd)

It work for me.
T31 tested on ESP8266-DEV Olimex

plinioseniore, 2015-09-18:
The C8TO16 takes two bytes and build a unit16_t, so it will use also the first byte of the setpoint input, that make no sense.

If this is working is because we have some case where with input == to RstCmd we still need to process the logic.

Any hint?

fazioa, 2015-09-18:
Also deleting line 122 it work!

plinioseniore, 2015-10-08:
@fazioa @mcbittech with the work on the thermostat you have deeply tested the T31, please let me know if this can be considered closed.

mcbittech, 2015-10-08:
For me must be tested more.
With thermostat we can test it deeply.
I say you when it can be closed.

Flavio.

juanpintom, 2015-10-21:
Hi all, I added a T31 and get an infinite loop on the log:
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB01><|0x23|0x17|0xAB01|0xAB05>
(MaCaco)<OUT><|0x15|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB01><|0x23|0x17|0xAB01|0xAB05>
(MaCaco)<OUT><|0x15|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB01><|0x23|0x17|0xAB01|0xAB05>
(MaCaco)<OUT><|0x15|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB01><|0x23|0x17|0xAB01|0xAB05>
(MaCaco)<OUT><|0x15|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
....
This causes Android App run slowly and the node doesnt respond well.

plinioseniore, 2015-10-22:
Seems that your node has always something to transmit, have you a deadband on the analog value that you are reading?

juanpintom, 2015-10-22:
I get the temp value from a Dallas sensor with importanalog function
without a deadband, can you help me to add a deadband?

Regards

plinioseniore, 2015-10-22:
Share your sketch

juanpintom, 2015-10-22:
It's already integrated on my v4 sketch:
https://github.com/juanpintom/Souliss_ESP_Examples/tree/master/SoulissDomo_Device_Example/Souliss_ESP_v4

You can found the lines related to the T31 at 196 and 245 on this file:
https://github.com/juanpintom/Souliss_ESP_Examples/blob/master/SoulissDomo_Device_Example/Souliss_ESP_v4/SetupAndLoop.h:

Regards

juanpintom, 2015-10-22:
I removed the Import Analog line and the loop isn't on the log, so the problem seems to be the deadband.

Regards

plinioseniore, 2015-10-24:
Can you please point me in the line where you are reading the analog value?

juanpintom, 2015-10-24:
https://github.com/juanpintom/Souliss_ESP_Examples/blob/master/SoulissDomo_Device_Example/Souliss_ESP_v4/SetupAndLoop.h#L275

plinioseniore, 2015-10-24:
My bad, the deadband is not in the sketch but in Typicals.h defined as Souliss_T3n_DeadBand. Try to increase this value.

juanpintom, 2015-10-24:
Hi, tested with 0.05 0.1 and 0.5, same result

Regards

juanpintom, 2015-10-25:
Tested with this folk with hysteresis and same result too:
https://github.com/stefanst/souliss

Regards

plinioseniore, 2015-10-25:
Doesn't see anything different in that fork

juanpintom, 2015-10-25:
He added the hyteresis value and some changes on t3n file:

https://github.com/stefanst/souliss/commit/826a3e60d829b49a8a181cc5eadd12695183a57a

plinioseniore, 2015-11-08:
The bugfix by @stefanst has been merged into the main friariello branch

plinioseniore, 2015-11-29:
@mcbittech @fazioa had a chance to try the last modification by @stefanst? Are things working properly with your thermostat?

fazioa, 2015-11-29:
Yes. It seems works.

closed 2015-11-29 by plinioseniore