souliss/souliss #180: Support for RGB WS2812 Indexable Leds with Effects

juanpintom, 2015-10-01:
Hi, I don't know if this enhancement must be opened here, on the Soulissapp side or both, I'll start here.

We can easily integrate the basic RGB functions for this leds using Neopixel library from Adafruit, but if we want to have awesome effects and others we need a "new" typical (I think we can use T16) and a new interface on SoulissApp (just a combobox with effects or similar)

This was suggested by Litris on Spanish forum:
https://groups.google.com/forum/#!topic/souliss-es/ya3akohGM1U

Here is a link with a sketch for ESP device, based on Arduino IDE and can be easily merged with a Souliss one.
http://www.instructables.com/id/Neopixel-ESP8266-and-Custom-3d-printed-holder/?ALLSTEPS
http://www.instructables.com/files/orig/FVV/596O/IEOP6G1X/FVV596OIEOP6G1X.ino

What do you think ?

juanpintom, 2015-10-01:
This is simple example to get it working, I'll work to make an sketch on Souliss to use it like a standard RGB led strip.
https://github.com/adafruit/Adafruit_NeoPixel/blob/master/examples/simple/simple.ino

plinioseniore, 2015-10-03:
@juanpintom what's your goal? I think that you can use T16 if you want to have something similar to a standard LED strip. I don't know which could be a dedicated typical for this.

juanpintom, 2015-10-05:
Yes and no, we can use a T16 to make an standard RGB strip with a WS2812,
but we can use this indexable led strips to make awesome effects, here you
can see an sketch created by MeroFilosofo, using a T15 and sending commands
via OH he have 30+ effects, here a list:
                                case  2: rainbow_fade(); break;
                                case  3: rainbow_loop(); break;
                                case  4: random_burst(); break;
                                case  5: color_bounce(); break;
                                case  6: color_bounceFADE(); break;
                                case  7: ems_lightsONE(); break;
                                case  8: ems_lightsALL(); break;
                                case  9: flicker(); break;
                                case 10: pulse_one_color_all(); break;
                                case 11: pulse_one_color_all_rev(); break;
                                case 12: fade_vertical(); break;
                                case 13: rule30(); break;
                                case 14: random_march(); break;
                                case 15: rwb_march(); break;
                                case 16: radiation(); break;
                                case 17: color_loop_vardelay(); break;
                                case 18: white_temps(); break;
                                case 19: sin_bright_wave(); break;
                                case 20: pop_horizontal(); break;
                                case 21: quad_bright_curve(); break;
                                case 22: flame(); break;
                                case 23: rainbow_vertical(); break;
                                case 24: pacman(); break;
                                case 25: random_color_pop(); break;
                                case 26: ems_lightsSTROBE(); break;
                                case 27: rgb_propeller(); break;
                                case 28: kitt(); break;
                                case 29: matrix(); break;
                                case 50: strip_march_ccw(); break;
                                case 51: strip_march_cw(); break;
                                case 88: new_rainbow_loop(); break;
                                case 98: demo_modeA(); break;
                                case 99: demo_modeB(); break;

Here the sketch:
https://groups.google.com/d/msg/souliss-es/ya3akohGM1U/HX2O8i_BBQAJ

And an Awesome video, photos and the Rule for OH:
https://groups.google.com/d/msg/souliss-es/ya3akohGM1U/1mer4ylaBgAJ

Regards

marksev1, 2015-10-05:
+1 on this feature :)

ribico, 2015-10-05:
I am waiting for some of these LEDs fro China... so.. +1

plinioseniore, 2015-10-06:
That's very nice, but I'm having hard time to think at this into a typical.
Is very tight to the desired behavior.

How a typical should look like? What should it do?

From Mobile.

marksev1, 2015-10-06:
Maybe Juan should be the right guy to answer this, so lets wait for him for his ideas. If I may just give idea on how I see it. For example "modes" of a house with tied animations, like "party" mode or "alarm" mode or maybe "wake up" mode :). I don't know if Juan had that in his mind :)

juanpintom, 2015-10-06:
Hi, on the example here
<https://groups.google.com/d/msg/souliss-es/ya3akohGM1U/HX2O8i_BBQAJ>
MeroFilosofo
uses a T15 but I think we just need a typical similar to T16 but with a
*combobox* with all the effects posted before. The codes are implemented on
the example and he works over OH.

*Note*: Maybe we need to work on the example because it uses many delay(20)
on it

What do you think?

Regards

plinioseniore, 2015-10-07:
This can be redesigned as general purpose combox, then the user will be able to have multiple selection in once.

I will forward this into SoulissApp and see what they say about.

marksev1, 2015-10-07:
A bit of off-topic -> Juan will you still add preliminary WS2812 support into the ESP sketch of yours? Maybe just color picker and stuff thats been working for now (or maybe even that spanish guy's animations stuff as an intermediate before stuff gets implemented into Souliss main code),

juanpintom, 2015-10-07:
Ok Dario.

juanpintom, 2015-10-07:
We can work on a simple example
<https://github.com/souliss/souliss/issues/180#issuecomment-144805293>based
on NeoPixel library to get it working with Souliss and a T16, we just need
to change the Output lines with Neopixel function and Souliss values
instead of AnalogWrite (Souliss Value)

Regards

marksev1, 2015-10-08:
Cool then I'll order few ws2812 leds :)

plinioseniore, 2015-11-29:
This can be of interest,
http://www.ebay.com/itm/LED-RingPixels-WIFI-WifiPixels-16-x-WS2812B-RGB-LEDs-ESP8266-/281786638390?hash=item419bcae036:g:6XsAAOSwjVVV4rWN

plinioseniore, 2015-11-29:
I'm closing this issue, these type of LED can be used combining mutliple Typical yet supported. There is no plan to add on SoulissApp a dedicated one, if it will, the code on the board side will be updated.

closed 2015-11-29 by plinioseniore

marksev1, 2015-11-29:
Do I understand you correctly, that you are saying, there are already Typicals in place that could support this kind of LED?

plinioseniore, 2015-11-30:
Not exactly, this LEDs has the same property of any other RGB led the only
difference is that you can set different effects.

So, you can create with a T14 buttons to select and effect and a T16 to
choose the color.

Dario.

marksev1, 2015-11-30:
Will you be playing with this leds? Can we expect a example in the souliss repo in the future :)

juanpintom, 2015-11-30:
Yes, effects can be done with T14, but we'll need 30+ T14 to have all the
effects :(  A combobox on a T16b with this effects will be better.

Regards

plinioseniore, 2015-12-01:
We could use a single button that changes the effect at every press. A
couple of WifiPixels are on the way, I will play a bit with.