souliss/souliss #185: My node is showing empty with ESP8266

terrag42, 2015-10-04:
Hello, today I incorporated the new Friariello library into my Arduino "libraries" folder. At first I had an issue with compiling, but [plinioseniore](https://github.com/plinioseniore) was very quick to fix it and issue a commit. Thank you! 
Now I am having trouble getting my nodes to show anything other than empty. I'm using the same code from issue #184, and I also tried [another sketch](http://hastebin.com/kiwitisaru.coffee) that I had working previously with a DHT sensor as a typical. Souliss app has no trouble connecting to the node, but for each sketch I try, the node is empty. Each time I upload a sketch, I tried dropping the DB and refreshing, and each time the app tells me that "1 Souliss nodes are stored in DB, totally 0 typicals are present"

plinioseniore, 2015-10-04:
@terrag42 the latest code on the branch may be unstable or under development, is there a specific reason why you are not using the latest public release?

https://github.com/souliss/souliss/releases

plinioseniore, 2015-10-04:
If you feel brave, you can pull again the latest commit, or skip and move to https://github.com/souliss/souliss/tree/v7.0-friariello.6

terrag42, 2015-10-04:
Thanks for the quick response AGAIN. I feel sheepish that I didn't know about the difference between the latest public release and the latest commit. Using the public release, my LEDs are updating really fast, and jammin' in time with the music. Thanks.

plinioseniore, 2015-10-04:
No problem, is just that we are used to publish code directly on the main branch when are minor modification and may happen that doesn't work properly. When a stable point has been reached, there is a public release that is a code that works fine for most of the cases.

From the actual main branch there will be the next release.

closed 2015-10-04 by plinioseniore