souliss/souliss #187: DONTROUTE prevent peers join

ribico, 2015-10-06:
Dear friends,
after my last update a big part of the network seems no more able to join the GW (from node number 5 on)
Reverting back to the last working setup (souliss a1e868f5 of 25 August) I got everything up again.

So the issue should be introduced after that commit.

Here are some logs from the latest commit (not working) and the a1e868f5 (working)
[not_working.txt](https://github.com/souliss/souliss/files/8772/not_working.txt)
[working.txt](https://github.com/souliss/souliss/files/8773/working.txt)

As Dario suggested it seems the not working setup logs DONTROUTE, could it be the origin for this issue ?

plinioseniore, 2015-10-06:
Likely, this commit f36f411658 fix your issue

ribico, 2015-10-07:
tested and working, thanks.

closed 2015-10-07 by ribico