souliss/souliss #201: T16 set Brightness issue

marcolino7, 2015-10-27:
Setting up brighness, the color of the led change, as per this post on ML:
https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!topic/souliss-it/EynQRqapQVo

marcolino7, 2015-10-29:
Hi Dario,
bottom to this sketch, you'll find a T16bis and it is the one I'm working on.
https://github.com/marcolino7/Souliss_ESP8266_RGB/blob/master/Souliss_ESP8266_RGB.ino

Setting Brigthness work for all colors and white
Do not work yet:
1) Fade IN and OUT when turn on and off
2) Flash button do nothing and I do not know how it should work
3) Sleep button do nothing and I do not know how it should work
4) I have not change to try music sync yet

Please have a look on my code and let me know

plinioseniore, 2015-11-08:
@marcolino7 have a try with 2b66ade61e and let me know if T16 behaves correctly

marcolino7, 2015-11-08:
Hi Dario,
I fixed compilation issue. Things does not work:
ON/OFF Button
Flash Button
Sleep Button

Brightness regulation have some issue. For example, if I setup yellow at max brightness, became red at minimum brightness

marcolino7, 2015-11-08:
Hi @plinioseniore 
all work. With last commit, fade out still not working. When you press off, wait a couple of second, then turn off led without fade

plinioseniore, 2015-11-10:
@marcolino7 try the last commit, fade-out should be ok now.

marcolino7, 2015-11-24:
Hi Dario
T16 still have this issue:

Ciao Dario,
I just loaded this sketch deom MeroFilosofo, 
it works also OTA.
Still remain issue on T16 just described 10 days ago in Chat 

1) Startup Node
2) Press ON and led strip turn on
3) Press OFF and led strip fade off
4) Press ON again, led turned on
5) Change the color, the black circle inside color circle still remain black
6) OFF does not work anymore

Marco

marcolino7, 2015-11-24:
In addition following command have no effect and do nothing:
FLASH
SLEEP

plinioseniore, 2015-11-29:
@marcolino7 commit 04b20b890 fix the problem with OFF command, FLASH is not used actually and will be added.

The SLEEP is based on the external timer, how you actual sketch use the timer?

closed 2016-01-16 by plinioseniore