souliss/souliss #205: Simple HTTP Interface

plinioseniore, 2015-10-28:
In the next SoulissApp releases Zozzariello will be removed, actually the HTTP and XML interfaces are not supported on ESP8266. The support should be extended to ESP8266 with a new interface, simpler than the actual XML and available for both AVR-W5x00 and ESP8266.

lulol, 2019-05-26:
The documentation and <a href="https://github.com/souliss/souliss/wiki/XML-Interface">wiki</a> doesn't mention anything about being deprecated.

What is the alternative to xml then? <a href="https://github.com/souliss/souliss/issues/318">#318</a>

Any hope to get back this functionality on the now ubiquitous ESP8266/ESP32?

plinioseniore, 2019-05-27:
The ESP is powerful enough to have Souliss running with other code at same
time, you can use any MQTT or HTTP library and build your set of commands
to interact with Souliss.