souliss/souliss #207: ESP8266 Mesh Network

crosofg, 2015-11-01:
Hi Can any sort of mesh networking be achieved using Souliss Framework?
Can you guys give something to start?

plinioseniore, 2015-11-01:
The ESP8266 cores for Arduino has example for WiFi mesh, I'm having a look
at to understand how it works and if can be used with Souliss

plinioseniore, 2015-11-01:
@crosofg had a quick look, the ESP8266 cores for Arduino support a very raw mesh that is basically a mix of STA and AP mode, so that nodes can interconnect but it doesn't offer any routing, so isn't a mesh at all.

Souliss doesn't have a mesh, but a tree structure and you can use it with a combination of STA (connection to an access point) and AP (access point), if you are interested to cover a longer range we can have a try.

crosofg, 2015-11-01:
Yeah longer range will be great. Can you have a look at this? 
https://github.com/espressif/ESP8266_LIGHT_WITH_MESH

crosofg, 2015-11-14:
Hi Did you get a chance to look the mesh implementation provided by Espressif?

plinioseniore, 2015-11-14:
You can try your self combining STA and AP mode to get your needs.

closed 2015-11-14 by plinioseniore

kubark42, 2018-02-09:
This does seem to be the killer application for this stuff. I'd like to have a home automation work on a MDU (multiple dwelling unit) and so mesh is critical to making it work since we don't have an in-building network.