souliss/souliss #208: [T3n] Need new Typical for Electric Heaters

YoannLe, 2015-11-01:
I'm working on design of boards for souliss based on ESP8266.
I've finish to test my version for electrics curtains controls and I will publish code and boards files soon.

Common goals for these designs:

the boards must fit in standard electric wall boxes 
* be supplied by 230V
* limit the number of components for DIY feasibility
* under 50€ by complete unit (logic and power board)
* common logic board and different power boards

I want to fork my power board to make a new board to control electric heater mode (eco, on, off...).

In France (and Europe) electrics heaters comes with 1 wire protocol called "Fil Pilote" and I need a new Typical to pilot this function. I'm not enough skilled to write this typical and I need help on this point.

Definition T33 Typical Electric Heater "Fil Pilote":

1 logical input with 6 states:

Confort (ON)
Confort -1°C (ON -1°)
Confort -2°C (ON -1°)
Eco (ON -3.5°)
Hors-Gel (Frost Protect)
Arrêt (OFF)
2 outputs:
Confort: O1=0 O2=0
Arrêt (OFF): O1=1 O2=0
Hors-Gel (Frost Protect): O1=0 O2=2
Eco: O1=1 O2=1
Confort -1°C (ON -1°): cyclic: 3 seconds O1=1 O2=1 / 297 seconds O1=0 O2=0 / 3 seconds O1=1 O2=1 / ....
Confort -2°C (ON -1°): cyclic: 7 seconds O1=1 O2=1 / 293 seconds O1=0 O2=0 / 3 seconds O1=1 O2=1 / ....

Each output will drive MOC3041 and the rectifier 1N4008 will provide combinaisons of waveforms.

[image]
[image]

plinioseniore, 2015-11-01:
Nice, does you need to synch with the zero crossing?

From Mobile.

plinioseniore, 2015-11-01:
Ok, looking at your circuit the first four are just a static set of two
bits that drives the mosfets.

What about the last two?

From Mobile.

YoannLe, 2015-11-01:
No need sync, modes are based on positive, negative or both part of AC waveform.Two optionals modes use a low timing (few seconds) alternate of no signal and full AC waveform.

 Envoyé le : Dimanche 1 novembre 2015 21h49
 Objet : Re: [souliss] [T3n] Need new Typical for Electric Heaters (#208)

plinioseniore, 2015-11-01:
Yes got it, we can use four T11 to set the operating modes and force the
bits that drives the gates of your mosftets.

Is your hardware ready?

YoannLe, 2015-11-03:
I've updated my schematic, PCB is almost routed, I will send it to the fab next days.

plinioseniore, 2015-11-08:
@YoannLe let me know when you are ready to test, in the while I'm closing this issue

closed 2015-11-08 by plinioseniore