souliss/souliss #209: T18 does not turn off relè if feedback not come back

marcolino7, 2015-11-07:
T18 does not turn off the control relè if feedback is not coming.
In case of failure of feedback circuit, the stepper relè may be burned.
A safe timer of some second should be implement to turn off relè in case feedback is not coming

plinioseniore, 2015-11-08:
Can you please post your sketch

marcolino7, 2015-11-08:
https://github.com/marcolino7/Souliss_Ingresso/blob/master/Souliss_Ingresso.ino

plinioseniore, 2015-11-08:
Looks that you are missing the timer,
https://github.com/souliss/souliss/blob/friariello/examples/DINo/e07_StepRelay/e07_StepRelay.ino

FulvioSpelta, 2015-11-09:
I confirm, the following part has to be added for every T18 (approx 500 ms timer);`

		FAST_x10ms(10) {

			// The output coil to the step relay is active for five cycle, so approximately five times 100 ms

			// use a different value to match your requirements

			Timer_StepRelay(STEPPER1);

		}

`

closed 2015-11-09 by FulvioSpelta