souliss/souliss #214: Support for new 2.0 ESP8266 cores

plinioseniore, 2015-11-29:
New ESP8266 cores has been relased in staging, support shall be extended. In the new cores:
1) A rewritten OTA support
2) PWM handled directly in RAM, this remove the needs for some tricky workarounds included in the LYT8266 examples.

plinioseniore, 2016-01-04:
The esp8266_core2 has started

juanpintom, 2016-01-04:
Im working with 2.0 and webupdate without problems.

PWM it's working well but I see blinks (bright changes) sometimes.

Regards

dariocdj, 2016-01-04:
Juan also with 1.6 with my dimmer sometimes it blinks one time....

Inviato da smartphone Samsung Galaxy.

-------- Messaggio originale --------

Data:04/01/2016  18:45  (GMT+01:00) 
A: souliss/souliss <souliss@noreply.github.com> 
Cc: Dario <cdj@deskservice.com> 
Oggetto: Re: [souliss] Support for new 2.0 ESP8266 cores (#214)

closed 2017-08-16 by plinioseniore