souliss/souliss #219: NRF Direct support for ESP8266

dariocdj, 2015-12-15:
The idea is to integrate in Souliss NRF Library for ESP8266. In this way we can build an integrated gateway (esp+nrf) that can support Low Power Batteries nodes. 
MySensors esp8266 gateway connect esp to nrf in this way : 
http://www.mysensors.org/build/esp8266_gateway

I've built (in my previous system) stable and cheap nodes with arduino pro mini 3.3v + nrf24l01 with some simple mod : removing regulator, removing power led, burn fuses and disable brown-out for operating at 1.8v. I've active nodes (simple reed) with 2 AA batteries running from 1,2 years !

Searching the web i've found Optimized High Speed NRF24L01+ Driver with ESP8266 support via Arduino IDE at http://tmrh20.github.io/RF24/index.html. Pheraps could be the solution ?

archiijs, 2015-12-16:
It would be weary nice, I still think of NRF node for my mailbox, so
integrated library wold make things easier for me.

plinioseniore, 2015-12-16:
I will, there is a yet ready library.

From Mobile.

dariocdj, 2015-12-16:
Thanks waiting 4 u ....

juanpintom, 2015-12-16:
+1, am not agree with nRF, was unstable for me, but I have nRF to test it.

Regards

marcolino7, 2015-12-16:
I leaved nfr due unstability caused by power supply.
I do not go over to test battery node. I have a lot of this modules, hope in future battery node live again.

Marco

dariocdj, 2015-12-16:
Marco, believe me, after 1 year (10mt distance) pheraps this node has crashed 1 TIME !!! IN 1 YEAR !!!! and i've used 2 AA Batteries that after 1 year are at 82% of charge (!!!!) 
Ah, is there planned to put charging level of batteries in nodes ?

pistoletov1974, 2015-12-16:
+1 nrf24l01 is realy nice modules!  Is it possible to make nodes on nrf24le1 (nrf+8051)?

marcolino7, 2015-12-16:
Hi Dario2,
me and Dario1 worked to Battery Node, but we were stopped by some issues. My problem with nrf is with node powered by 220 Volt and power supply, they are more unstable when power is not filtered. For this reason I abandoned nrf for ESP into 220V powered node.
If you have experience with nrf and battery node, please share your experience on ML, I'm very intrested.

Marco

dariocdj, 2015-12-17:
Absolutely, if you want when nrf24 esp support is ready i can write a tutorial to build low power node, of course... Waiting for plinioseniore porting  library in souliss... Dario1 we're waiting you :)

pistoletov1974, 2015-12-17:
My modules nrf did not work without electrolytic capacitor on power pins. I strongly recomend to use it with 220v or battery supply. Few evenings i thought that problem was  in code but after adding capacitor all problems have been resolved.

marcolino7, 2015-12-17:
@pistoletov1974 I know about capacitor, but I have an old house with thick walls, and gateway is on another floor, so some distance nrf node loose connection with gateway. For 220V powered node I moved to ESP, but I'm looking to a valid solution for Battery node too.

Marco

dariocdj, 2015-12-17:
In my old sys it works perfect without capacitor... the only electronic component is 1M resistor for reed (door/window low power sensor).

plinioseniore, 2015-12-17:
A preview will be available late today, I will just verify that compiles properly with a dedicated example. Actually I cannot do further testing, so please keep you boards ready for testing.

dariocdj, 2015-12-17:
I'm ready, if you can give us connections we can prepare hardware....

plinioseniore, 2015-12-17:
Same as MySensors

| NodeMCU|Radio|Comment|
|:---:|:---:|:---:|:---:|
|GND|GND|Black|
|3V3	VCC|Red|
|D2|CE|Orange|
|D8|CSN/CS|Yellow|
|D5|SCK|Green|
|D7|MOSI	Blue|
|D6|MISO|Violet|

http://www.mysensors.org/build/esp8266_gateway

dariocdj, 2015-12-17:
Ok for people without nodemcu : 
 * nRF24L01+  ESP8266
 * VCC        VCC
 * CE         GPIO4          
 * CSN/CS     GPIO15
 * SCK        GPIO14
 * MISO       GPIO12
 * MOSI       GPIO13
 *

plinioseniore, 2015-12-17:
@dariocdj have a try with [this](https://github.com/souliss/souliss/tree/nrf24_esp8266) branch that is more or less a copy/paste of the code @TMRh20. I haven't done any test, I just now that this compile. 

There are two example that could be used for the scope, 
[e05_WiFi_nRF24_gateway](https://github.com/souliss/souliss/tree/nrf24_esp8266/examples/wireless/e05_WiFi_nRF24_gateway)
[e05_WiFi_nRF24_peer](https://github.com/souliss/souliss/tree/nrf24_esp8266/examples/wireless/e05_WiFi_nRF24_peer)

Let me know if you get it working.

plinioseniore, 2015-12-17:
Just seen that I've forgot to add the following:

```C
#ifndef NRF24PINS_INSKETCH
#define NRF24_RADIOEN       4           // Chip Enable Pin
#define NRF24_SPICS           15           // SPI Chip Select Pin
```

This will define the CE and CS pin used for the scope, I will add those by default while using the ESP8266 if things will work.

dariocdj, 2015-12-17:
Ok, i think it miss #endif... but what library do i must install ? the one found in http://tmrh20.github.io/RF24/index.html ?
because - obviously - with normal #include "conf/nRF24L01.h"  there's a tons of compile error....

Tnx
Dario

plinioseniore, 2015-12-17:
@dariocdj the libraries are included in Souliss and you don't need addition libraries.

It compile succesfully with me, have you installed code from the new branch linked before?

plinioseniore, 2015-12-17:
Please note that is #define and not #ifndef alse I'be wrongly wrote before.

dariocdj, 2015-12-17:
Ok sorry, didn't download new branch, it compile without errors.... i try gateway....

plinioseniore, 2015-12-17:
Let me know if you get it working

dariocdj, 2015-12-17:
Ok guys, esp working with another souliss sketch, also with this new branch... when i upload e05_WiFi_nRF24_gateway esp restart in loop :(without nrf connected...) :+1: 
wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
 

wtf ?

plinioseniore, 2015-12-17:
Same if you have the nRF connected? It may be that is looking for the device on the SPI and get in stuck while waiting for an answer.

I've not looked deeply in the code, but your ESP8266 is having a watchdog reboot, that is usual when the code is locking.

dariocdj, 2015-12-18:
Sorry, i've tried also with nrf connected and double checked connection but always similar result : 

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Waiting 4 u...

dariocdj, 2015-12-18:
meanwhile, Just to do something, i'm programming arduino pro mini with peer sketch and connection of nrf using this scheme : 

nRF24L01+      PRO MINI 3.3V
GND                   GND
VCC                   VCC
CE                      D9
CS                     D10
SCK                  D13
MOSI                D11
MISO               D12

Programmed regulary... but can't test it ....

dariocdj, 2015-12-18:
OK NEWS : 
Missing define must be put in code immediately after 
#define WIFICONF_INSKETCH
#define WiFi_SSID               "mywifi"
#define WiFi_Password       "myfpass"   

Ok now gateway is online.... let me test if peer works...

dariocdj, 2015-12-18:
Woww... pheraps something is missing, but SoulissApp recognize 2 nodes (!!!) but 0 devices inside nodes....  So, if nodes is defined on Arduino+NRF peer ( Set_T11(LIGHT1_NODE1);
Set_T11(LIGHT2_NODE1); ) it would tell us that NRF BRIDGE WORKS !!! Or i'm understanding wrong something?

marcolino7, 2015-12-18:
Hi Dario,
souliss app download node from Gateway and Typical from nodes. So if you get Empty nodes perhaps nrf comunication is not working

dariocdj, 2015-12-18:
Yes that's it... two nodes found are Gateway itself and nrf bridge i suppose (0x6501)....
and then, how to debug this issue ?

marcolino7, 2015-12-18:
We need Dario1 @plinioseniore

plinioseniore, 2015-12-18:
Enable the debug on both nodes and include here the results.

dariocdj, 2015-12-18:
Ehm, sorry for question, how i can enable debug? #define DEBUG? And then output of serial?

plinioseniore, 2015-12-18:
Here the details, logs will be printed on the Serail monitor https://github.com/souliss/souliss/wiki/TroubleShootSouliss

dariocdj, 2015-12-19:
Ok li attivo tutti? O basta vnet e souliss?

Inviato da smartphone Samsung Galaxy.

-------- Messaggio originale --------

Data:18/12/2015  23:52  (GMT+01:00) 
A: souliss/souliss <souliss@noreply.github.com> 
Cc: Dario <cdj@deskservice.com> 
Oggetto: Re: [souliss] NRF Direct support for ESP8266 (#219)

dariocdj, 2015-12-19:
Ops sorry for Ita, ok can I turn on all logs or it's sufficient only vnet and souliss one?

plinioseniore, 2015-12-19:
vNet is enough in this case. Thanks.

From Mobile.

dariocdj, 2015-12-19:
Here's the log... but seems to be all 0 ....

[log_gw.txt](https://github.com/souliss/souliss/files/67502/log_gw.txt)
[log_peer.txt](https://github.com/souliss/souliss/files/67503/log_peer.txt)

plinioseniore, 2015-12-19:
Thanks @dariocdj seems that something is transmitted or that drivers gets wrongly data available flags, even if there is nothing more that a bunch of zeros.

I will try to have a look at, but I don't know if it will be easy without having the hardware with me.

dariocdj, 2015-12-19:
do you want me sending a couple of nrf?

Inviato da smartphone Samsung Galaxy.

-------- Messaggio originale --------

Data:19/12/2015  16:12  (GMT+01:00)

plinioseniore, 2015-12-19:
I've some nRF radios as well as some ESP8266, but not time to install the nodes and test them in these days. I may try to have a look at the code, to see if something is odd and issue a new code to be tested.

Otherwise I will setup a demo to test it out as soon as I will have some time.

dariocdj, 2015-12-22:
@plinioseniore do you think that it's better than i start with an esp-arduino-nrf gateway or pheraps in this holiday you can try to fix it?

plinioseniore, 2015-12-29:
@dariocdj sure you can use the nRF connected to an AVR, that will work. I had not time to look at the code and have a side by side comparison.

plinioseniore, 2017-08-16:
Deprecated

closed 2017-08-16 by plinioseniore