souliss/souliss #220: Arduino MEGA2560+ENC28J60...

DD-WRT, 2015-12-17:
Greetings to everyone ,
I do the electronic, I built souliss with ATMEGA 2560 + ENC28J60+Nrf24L01+PA as the gateway, and the Arduino mini pro + NRF24L01 node, as I put the sketch hello world in the mega 2560 and have established communication with APP, works great run relay  on-off,  will now issue is that sketch to enroll in gateway that the node or set to match them in the system.
* I need a control 10-15 nodes and one gateway, and the ability to measure temperature at one node.
Thank you all for help

plinioseniore, 2015-12-17:
Isn't much clear your problem, you can't get your Peer node online?

DD-WRT, 2015-12-17:
I got online gateway(souliss-app go green and respond), but I have only Mega + ENC, Nrf is not connected and I do not know how to set up the sketch, because it only works on gateway relay (use only for test) have constructed and RX-node but do not have a sketch for but do not know how to sign on gateway.

plinioseniore, 2015-12-17:
In the folder example/wireless there are several examples, the 02 is a good starting point. Consider to have proper connections,
https://github.com/souliss/souliss/wiki/Transceiver%20Connection

DD-WRT, 2015-12-17:
plinioseniors, first to thank you for the answer, if he can ask that we propose finished .ino file that needs to put in gateway and also .ino file that needs to put in the node. Hardware, electronics I know a lot but this is the writing sketch and libraries arduino not much so please if you can give me sent file arranged to throw and get it to run as an example.
Thanks again

plinioseniore, 2015-12-17:
The given example should be a good starting point, use those for your case and post your sketches if you didn't get them working.

DD-WRT, 2015-12-20:
Thanks a lot for you,  this works great  :)
I still question what to change to work with ENC28J60 ,
I tried to change  #include "conf / ethW5100.h"  to this  #include "/conf/ethENC28J60.h" but ide 
tells me error can not find my library , 
and how to add more peer nodes (I think you only need to change the #define Peer_address) and maybe I'm wrong.
Thanks a lot

plinioseniore, 2015-12-24:
This is an example for ENC28J60 https://github.com/souliss/souliss/blob/friariello/examples/ethernet/e01_HelloWorld_uIP/e01_HelloWorld_uIP.ino

DD-WRT, 2015-12-24:
Thanks Dario ,
I tried it and it works, but in there skech is no support for the Nrf module, I see that when using ENC28J60 uses the same bus in which a real conflict of MISO, MOSI line and the question is how to assign other software SPI pins for NRF modules.
We all excellent works with W5100 network module (successfully connected, nodes, gateway, relay, but I want to connect with ENC.

plinioseniore, 2015-12-24:
You can use the SoftwareSPI incldued in DigitalIO library, you need to download that library and follow the istruction to set it up with Souliss
https://github.com/souliss/souliss/wiki/Transceiver%20Connection#nordic-nrf24l01-and-nrf24l01

closed 2016-01-23 by plinioseniore