souliss/souliss #225: Error compliling Souliss-HelloWorld example.  Please Help.

YTNic, 2016-01-10:
Hello,
I am fairly new to Arduino in general, but i have had several projects up and running so I do understand most of the basics.  I am just now starting to work with Souliss and I can't even get the Ethernet-HelloWorld example to run.  I get the message "fatal error: bconf/StandardArduino.h: No such file or directory".  I have not added or changed anything in the example and the bconf file is there.  I am simply loading it from the examples menu and then when I go to verify, this is what happens every time.  I have been searching all over for a solution, but I have found nothing.  

Here is the full error message...

Arduino: 1.6.7 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\JNB\Documents\Arduino\libraries" -fqbn=arduino:avr:mega:cpu=atmega2560 -vid-pid=0X2341_0X0042 -ide-version=10607 -build-path "C:\Users\JNB\AppData\Local\Temp\build356412532eb7a685ca3c79576be676ec.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\JNB\Documents\Arduino\libraries\souliss\examples\ethernet\e01_HelloWorld\e01_HelloWorld.ino"
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\JNB\Documents\Arduino\libraries" -fqbn=arduino:avr:mega:cpu=atmega2560 -vid-pid=0X2341_0X0042 -ide-version=10607 -build-path "C:\Users\JNB\AppData\Local\Temp\build356412532eb7a685ca3c79576be676ec.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\JNB\Documents\Arduino\libraries\souliss\examples\ethernet\e01_HelloWorld\e01_HelloWorld.ino"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\JNB\AppData\Local\Temp\build356412532eb7a685ca3c79576be676ec.tmp\sketch\e01_HelloWorld.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\JNB\AppData\Local\Temp\build356412532eb7a685ca3c79576be676ec.tmp\sketch\e01_HelloWorld.ino.cpp" -o "C:\Users\JNB\AppData\Local\Temp\build356412532eb7a685ca3c79576be676ec.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
C:\Users\JNB\Documents\Arduino\libraries\souliss\examples\ethernet\e01_HelloWorld\e01_HelloWorld.ino:19:70: fatal error: bconf/StandardArduino.h: No such file or directory

 #include "bconf/StandardArduino.h"          // Use a standard Arduino
                                                                      ^
compilation terminated.

exit status 1
Error compiling.

Please help!!!  What am I doing wrong?
Thank you!!!

plinioseniore, 2016-01-10:
You are missing the dot
#include "bconf/StandardArduino.h"

I see that you are also using an unsupported release of the IDE, keep it in
mind if you will have further problems.

YTNic, 2016-01-10:
Thank you for your help plinioseniore.

I apologize, as I am very new to all of this.  I don't know what you meant by missing the "dot".  #include "bconf/StandardArduino.h"  is exactly how mine reads.  Am I just completely misunderstanding what you are saying?

As far as the IDE release, I didn't realize that it was unsupported.  I will try rolling it back and see if that helps.

Thank you again!

plinioseniore, 2016-01-10:
I'm sorry, on my mobile it was showing the error message with the dot before the "h", I haven't used Arduino IDE 1.6.7 and so I don't know if it doesn't search properly within the files.

How have you installed Souliss library? Through the Library Manager?

YTNic, 2016-01-10:
No worries on the dot issue.

Yes, the Souliss library is installed and I did install it through the Library Manager.  I have also tried removing it and reinstalling it as a .zip, but neither way has worked.

I think you may be on to something with IDE version.  

Should I just try 1.6.6 or something other than that?

plinioseniore, 2016-01-10:
IDE 1.6.5 https://github.com/souliss/souliss/wiki/Supported%20Hardware#supported-cores

plinioseniore, 2016-01-10:
By the way, I'm downloading the 1.6.7 to have a try

YTNic, 2016-01-10:
Very good I will give it a shot.

Thank you so much for your help!!!

plinioseniore, 2016-01-10:
I can confirm that in IDE 1.6.7 has been changed the pre-processor and it will not find the configuration files, move to 1.6.5 to get things working, in the while I will look for a solution.

closed 2016-01-10 by plinioseniore

YTNic, 2016-01-10:
I am now using 1.6.5 and it is working perfectly.

Being new to all of this, I just assumed that I was doing something wrong.
I'm so glad that it wasn't just me.

Huge thanks again for taking the time to help out the "new kid"!  You have
really made me feel very welcomed in a world that is very new to me!

-Nic

zoomx, 2016-09-24:
Got the same error, e01_HelloWorld example for Arduino UNO and IDE 1.6.11 (core 1.6.13), library installed with library manager. The same with IDE 1.6.9 (core 1.6.11). Library version is 7.1.1
Reading here
https://github.com/arduino/Arduino/issues/4407
putting #include "Souliss.h" as first you get some others error (multiple definitions)
but creating an empty SoulissBase.h in Souliss library folder and putting #include <SoulissBase.h> as first works.
I believe that, as stated on issue 4007, that the library should move to a src subfolder.

plinioseniore, 2016-09-25:
In the wiki there are the supported IDEs and cores.

zoomx, 2016-09-25:
Wiki says >=1.6.5 for IDE and cores
[https://github.com/souliss/souliss/wiki/Supported%20Hardware#supported-cores](url)

plinioseniore, 2016-09-27:
@zoomx I've updated the wiki, you need to use 720_rc branch in order to compile with IDE >=1.6.7 where is included a dedicated header file as you stated before.