souliss/souliss #226: Arduino IDE 1.6.7 New Preprocessor doesn't compile with Souliss

plinioseniore, 2016-01-10:
In the new IDE the preprocessor has changed and the configuration files are not recognized by the IDE

plinioseniore, 2016-01-10:
For people interested, follow the discussion here
https://github.com/arduino/Arduino/issues/4407

plinioseniore, 2016-01-11:
To fix it, as suggested by Arduino team is add a blank .h file at the beginning, this will direct the IDE in the right direction.

IT9IPQ, 2016-01-12:
To confirm, with the ide 1.6.7 and souliss-7.1-friariello.1 I got the correct (no library error) compilation by creating an empty file SoulissBase.h into a external library (where Souliss.h) and entering in the sketch at the beginning as the first library the line:
` #include <SoulissBase.h> // load patch to the library for IDE 1.6.7 `

plinioseniore, 2016-01-23:
I've updated the examples to compile for 1.6.7, this applies only for AVR based examples because the ESP8266 cores seems not yet ready for 1.6.7.

In any case, the supported IDE is 1.6.5 and will not support officially 1.6.7 but likely just to a new release once ESP8266 core 2 will be full functional with IDE.

plinioseniore, 2016-02-05:
Wiki to be updated

fazioa, 2016-02-23:
+1
Thanks

plinioseniore, 2020-01-06:
Wiki seems fine

closed 2020-01-06 by plinioseniore