souliss/souliss #229: Arduino + ESP + NRF Routing Problem

dariocdj, 2016-01-21:
Hi everybody.
There's a bug in routing if you have an ESP gateway with an Arduino Supernode USART connected and an NRF24L01  Receiver. 
You must add a  RoutingTable(0xd102, 0x6500, 0); to gateway or your peer NRF is not recognized.

Dario Cdj

plinioseniore, 2016-01-21:
Actually the ```vNet_GetSourceAddress_M2()``` and ```vNet_GetSourceAddress_M5()``` returns always zero, because are both based on a broadcast media that doesn't trasfer by default the address of the sender. So is not possible to identify the middle step between each routed frame.

This can be fixed at vNet driver level, adding as info the address of the sender of a certain frame, because in the vNet frame itself there is only the starting sender and the final destination address, nope all the middle step in between. This has been done for M3.

As is now, the only vNet drivers that allow an autodiscovery of paths are M1 and M3.

plinioseniore, 2016-01-21:
I've updated the wiki to specify that only for some cases the automatic path detection works
https://github.com/souliss/souliss/wiki/Add-Multiple-Media#automatic-path-detection

closed 2016-11-15 by dariocdj