souliss/souliss #236: Temperature Node

DD-WRT, 2016-02-05:
Hello,
I made souliss  system with the gateway i 10 on / off nodes and works perfectly,now the question is how to make temperature (DS1820 or LM35 ) sensor node, right can work wirelessly with NRF24L01 and Arduino UNO/ Mini ??
Thanks :)

stefanst, 2016-02-05:
Running a Dallas 18b20 is fairly simple. Look at code below:

#define DALLAS               1              // Address for temp logic

#define DALLASPIN            3              // Dallas Temp sensor is hooked up here. Could be any digital pin
OneWire ourWire(DALLASPIN);                 // Create an instance of OneWire using the pin defined above
DallasTemperature sensors(&ourWire);        // create an instance of DallasTemperature

void setup() {   
  Set_Temperature(DALLAS);
  ...
}

void loop () {
  .....
    FAST_910ms()    {
      sensors.requestTemperatures(); //Get the sensor ready for reading
      float dallas = sensors.getTempFByIndex(0);
      Souliss_ImportAnalog(memory_map, DALLAS, &dallas);
....

DD-WRT, 2016-02-07:
stefanst ;
 thanks for your reply, I found skech with DHT support for a power socket just to get DHT sensor and try if it worked.

closed 2016-02-07 by DD-WRT