souliss/souliss #248: DHT11 ,not show temperature on app ?

DD-WRT, 2016-03-24:
I have a problem with displaying temperature, humidity conditioning works great :)

[image]

#include "DHT.h"

// Configure the framework
#include "bconf/StandardArduino.h"          // Use a standard Arduino
#include "conf/nRF24L01.h"
// Include framework code and libraries
#include "SPI.h"
/*** All configuration includes should be above this line ***/
#include "Souliss.h"

// Uncomment whatever type you're using!
#define DHTTYPE DHT11   // DHT 11

#define DHTPIN 2     // what digital pin we're connected to

DHT dht(DHTPIN, DHTTYPE);

// Define the network configuration according to your router settings
#define Gateway_address 0x6501              // The Gateway node has two address, one on the Ethernet side
// and the other on the wireless one
#define Peer_address3    0x6503
#define myvNet_subnet   0xFF00
#define myvNet_supern   Gateway_address
#define slot_dhtt    1
#define slot_dhtu    2

void setup() {

  Serial.begin(9600);
  Serial.println("DHTxx test!");
  dht.begin();
  // Define inputs (hardware pulldown required) and outputs pins
  pinMode(2, INPUT);                 // Power to relay coil for
  Initialize();
  // Set network parameters
  Souliss_SetAddress(Peer_address3, myvNet_subnet, myvNet_supern);                 // Address on the wireless interface
  // Set the typical logic to handle the lights
  Set_T52( slot_dhtt );
  Set_T53( slot_dhtu );
}

void loop() {

  // Here we start to play
  EXECUTEFAST() {
    UPDATEFAST();

    FAST_2110ms() {
      Logic_T52(slot_dhtt);
      Logic_T53(slot_dhtu);

    }
  }
  // Here we process the communication
  FAST_PeerComms();

  SLOW_10s() {
    // Reading temperature or humidity takes about 250 milliseconds!
    // Sensor readings may also be up to 2 seconds 'old' (its a very slow sensor)
    float h = dht.readHumidity();
    // Read temperature as Celsius (the default)
    float t = dht.readTemperature();
    // Read temperature as Fahrenheit (isFahrenheit = true)
    float f = dht.readTemperature(true);
    // Check if any reads failed and exit early (to try again).
    if (isnan(h) || isnan(t) || isnan(f)) {
      Serial.println("Failed to read from DHT sensor!");
      return;
    }
    // Compute heat index in Fahrenheit (the default)
    float hif = dht.computeHeatIndex(f, h);
    // Compute heat index in Celsius (isFahreheit = false)
    float hic = dht.computeHeatIndex(t, h, false);
    ImportAnalog(slot_dhtt, &t);
    ImportAnalog(slot_dhtu, &h);
    Serial.print("Humidity: ");
    Serial.print(h);
    Serial.print(" %\t");
    Serial.print("Temperature: ");
    Serial.print(t);
    Serial.print(" *C ");
    Serial.print(f);
    Serial.print(" *F\t");
    Serial.print("Heat index: ");
    Serial.print(hic);
    Serial.print(" *C ");
    Serial.print(hif);
    Serial.println(" *F");
  }
}

plinioseniore, 2016-03-25:
Use two different slots, the DHT temp/hum can be read once per time if I
remeber correctly.

Dario.

From Mobile.

mcbittech, 2016-03-25:
And you have a wrong configuration of your slot:
#define slot_dhtt 1
#define slot_dhtu 2
must be:
#define slot_dhtt 0
#define slot_dhtu 2

T52 an T53 require 2 slot for each.

You can see here <https://github.com/souliss/souliss/wiki/Typicals>slot
occupation for each typicals.

Flavio.

DD-WRT, 2016-03-25:
Works excellent, 
I've changed slots , I did not know to use two slots :)
#define slot_dhtt  1
#define slot_dhtu 3
[image]

Thanks to Dario and Flavio .

closed 2016-03-25 by DD-WRT