souliss/souliss #249: fix crash on use ESP8266 WEBCONFinterface

setar, 2016-04-07:
interfaces/mcu_esp8266/webconfig/src/netconfig.h

- server.send (200, "text/html", reinterpret_cast<const __FlashStringHelper *>(PAGE_WaitAndReload ));
+ server.send_P (200, "text/html", reinterpret_cast<const __FlashStringHelper *>(PAGE_WaitAndReload ));

- server.send ( 200, "text/html", PAGE_NetworkConfiguration );
+ server.send_P ( 200, "text/html", PAGE_NetworkConfiguration );

P.S. http://www.esp8266.com/viewtopic.php?f=8&t=4685

plinioseniore, 2016-04-07:
@setar could you please make a pull request with the modified file?

Thanks.

plinioseniore, 2016-09-28:
@jkutianski could you please have a check and pull the modified files into 720_rc?

Thanks,
Dario.

jkutianski, 2016-09-28:
Dario.

I'm working on it
Best.

Juan Pablo

jkutianski, 2016-09-29:
@setar could you please test it?

jkutianski, 2016-10-01:
I will close the issue. If the problem persist, please reopen it.

closed 2016-10-01 by jkutianski