souliss/souliss #252: Value from Typical T58 (pressure sensor) is not updated

artur3, 2016-04-12:
Value from Typical T58 (pressure sensor) is not updated (constant pressure is shown) in soulissApp and openHaB. Resetting board usually solves the problem for short time.
Value printed to serial port is correct and changes over the time.
Other sensors are working fine.

Code (Arduino Mega + eth shield):
Setup
`  // Set the typical to use
  Set_T52(TEMP1);
  Set_T53(HUMIDITY);
  Set_T54(LIGHT);
  Set_T58(PRESSURE);
  Set_T52(TEMP2);`

Loop
   ` FAST_11110ms()
    {
      float temp1 = dht.readTemperature();
      ImportAnalog(TEMP1, &temp1);

      float humidity = dht.readHumidity();
      ImportAnalog(HUMIDITY, &humidity);

      float temp2 = bmp.readTemperature();
      ImportAnalog(TEMP2, &temp2);

      float pressure = bmp.readPressure();
      pressure = pressure / 100;
      ImportAnalog(PRESSURE, &pressure);

      float lux = lightMeter.readLightLevel();
      ImportAnalog(LIGHT, &lux);

#if defined(_DEBUG_)

      Serial.println();
      Serial.print ("Temp 1 = ");
      Serial.print (temp1);
      Serial.println(" *C");
      Serial.print ("Hum = ");
      Serial.print (humidity);
      Serial.println(" %");
      Serial.print("Temp 2 = ");
      Serial.print(temp2);
      Serial.println(" *C");
      Serial.print("Pressure = ");
      Serial.print(pressure);
      Serial.println(" hPa");
      Serial.print("Light = ");
      Serial.print(lux);
      Serial.println(" lx");
      Serial.println();

#endif

      // Compare the acquired input with the stored one, send the new value to the
      // user interface if the difference is greater than the deadband

      Logic_T52(TEMP1);
      Logic_T52(TEMP2);
      Logic_T53(HUMIDITY);
      Logic_T54(LIGHT);
      Logic_T58(PRESSURE);

      // Retreive data from the MaCaco communication channel
      ProcessCommunication();
    }
`

plinioseniore, 2016-04-17:
Data are transferred with a deadband, so if you are below that deadband nothing is transferred. Actually is 1.5% of last visualized value.
https://github.com/souliss/souliss/blob/friariello/base/SpeakEasy.h#L211

artur3, 2016-04-17:
You're right, 0,015 from thousands is more than usually pressure changes.
Maybe default deadband for pressure should be lowered?

plinioseniore, 2016-04-17:
What's the resolution of your sensor compared to the full range? Better that 1.5%?

artur3, 2016-04-17:
For example, serial port shows today 992 but SoulissApp still shows 998 from yesterday. Charts for pressure are flat line for days.  998 * 0.015 = 14,97

artur3, 2016-04-17:
BMP180
Up to 0.03hPa / 0.25m resolution
-40 to +85°C operational range, +-2°C temperature accuracy

plinioseniore, 2016-04-17:
Lower your value to 0.0015 and let me know if you network slows down due to an excess of data notified.

artur3, 2016-04-17:
Is the amount of data does not depend on how often I read sensors? Even every few minutes or less should be enough. 
Problem is 1.5% from thousands, that's more than usually pressure changes, so far my readings shows about 994 to 1015hPa.

plinioseniore, 2016-04-17:
Yep, you can slow down you measure read to be sure that it will not send data too much data. Consider that a node works reliably at an average of 1 update/s rate.

This is conservative value and the real one depends on your network structure and nodes.

closed 2017-08-16 by plinioseniore