souliss/souliss #253: Typical T16 is not reporting data change

artur3, 2016-04-12:
`if (Logic_T11(LIGHT))
  {
...
}`  
works when data is changed. Same code for T16 is not working.

plinioseniore, 2016-04-17:
That's strange, the T16 is supposed to transfer if a data has changes in case of ON/OFF and not in case of change in bright or color.

artur3, 2016-04-17:
Yesterday I found that this work for On/Off. But would be great to have this also for color and bright. So the command for the lamp can be send only when needed.

plinioseniore, 2016-04-17:
The main problem is that this will create trouble when synchronizing with music. You can any how change your own code to get this adding the i_trigger code for the ifs that you need.

Let me know.

artur3, 2016-04-17:
Ok. I can live with that.

closed 2016-04-17 by plinioseniore