souliss/souliss #260: My nodes

DD-WRT, 2016-04-29:
Hi all,
I have a question for someone who has built RGB LED strip, if you can send me skech to work as a node for NRF24 ??
I send my homemade node working perfectly...  :)
[image]

plinioseniore, 2016-05-01:
You got your node in the network, just add the relevant typical. In the
tutorial there is a sketch for the H801 that can be a starting point.

DD-WRT, 2016-05-01:
I'm sorry, if he can help me a bit more me exactly what you mean H801, I looked T16 al is to think that working as a special module with in 5100 is not what we should be me.
Thank you

ID-Electronic, 2016-12-11:
Hello everyone,
 I want to do RGB light or do not know how to connect my node to gateway, I have built over NRf24 nodes that work ON-OFF, temperature, I do not know how to write skech(T15,T16) for RGB light, I saw In the stand-alone module who works on the network, but I have a gateway. I want him to connect to Wi-fi node and work as whole RGB light.
I looked e03_RGBStrip.ino but how to set to work on my node as the NRf24 connected to the my gateway.
Thank you

closed 2017-08-16 by plinioseniore