souliss/souliss #261: Pushetta con MEGA

felicevitiello, 2016-05-01:
Pushetta su Arduino MEGA con la seguente configurazione si avvia:

#include "bconf/StandardArduino.h"          // Use a standard Arduino
#include "conf/ethW5100.h"                  // Ethernet through Wiznet W5100
#include "conf/Gateway.h"                   // The main node is the Gateway, we have just one node
#include "conf/Webhook.h"                   // Enable DHCP and DNS

.....

uint8_t ip_address[4]  = {192, 168, 177, 51}; //indirizzo peer corrente
uint8_t subnet_mask[4] = {255, 255, 255, 0}; //subnet
uint8_t ip_gateway[4]  = {192, 168, 177, 1}; // indirizzo gateway internet
uint8_t gatewaysouliss[4]  = {192, 168, 177, 51}; // indirizzo gateway Souliss
....

void setup()
{   
    Serial.begin(9600);
    Serial.println("Avvio");
    Initialize();
  
    // Network configuration
    Ethernet.begin();
    SetIPAddress(ip_address, subnet_mask, ip_gateway);
    SetAsGateway(myvNet_gateway);   
...
 }

Mentre con la seguente configurazione non funziona:

#include "bconf/StandardArduino.h"          // Use a standard Arduino
#include "conf/ethW5100.h"                  // Ethernet through Wiznet W5100
#include "conf/Gateway.h"                   // The main node is the Gateway, we have just one node
#include "conf/Webhook.h"                   // Enable DHCP and DNS

....

#define ip_gateway_address 1
#define gateway_address 51
#define myvNet_address  51
#define Peer_address_2  52
#define Peer_address_3  53
#define Peer_address_4  54
#define Peer_address_5  55

...

IPAddress ip(192, 168, 177, myvNet_address);
IPAddress gateway(192, 168, 177, ip_gateway_address);
IPAddress subnet(255, 255, 255, 0);

...

void setup()
{   
    Initialize();
  
    // Network configuration
    Ethernet.begin(ip, gateway, subnet);    // If DHCP fails, use a manual address                      
    SetAsGateway(gateway_address);       // Set this node as gateway for SoulissApp  

....
 }

Apparentemente la cosa mi sembra uguale, ma in realtà con la seconda versione pushetta non va...

tonypedi, 2016-05-02:
https://www.arduino.cc/en/Reference/EthernetBegin

hai omesso il mac address come primo parametro della ethernet begin

inoltre il parametro DNS, per fargli considerare il default, devi cmq inserire la virgola ( ipotizzo) , altrimenti lui considererà DNS ciò che tu chiami gateway ... sempre dopo che avrai messo anche il MAC all'inizio

adesso lui considera "ip" come se fosse il mac address, "gateway" lui lo considera ip, "subnet" lui lo considera dns

tonypedi, 2016-05-03:
1) forse in souliss è stata 'inglobata' una ethernet.begin che supporta SOLO la chiamata senza parametri e bisogna usare necessariamente la chiamata "IPAddress" per fornire gli altri parametri (noi in souliss usiamo la WEBHOOK e non le ethernet.h , ethernetudp.h)

2) cmq è ipotizzabile che anche facendo riferimento a https://www.arduino.cc/en/Reference/EthernetBegin negli sketch che fanno uso di ethernet.h, tu debba inserire delle virgole dove ci sono i parametri che tu hai omesso MAC e DNS, come la scrivi tu ... ciò che chiami ip lui lo interpreta come MAC, ciò che tu chiami gateway, lo lo interpreta come ip ... e via discorrendo

felicevitiello, 2016-05-03:
Appena ho tempo ci guardo meglio, potrebbe essere anche correlato ai
problemi che ho con adafruit.io, in effetti non avevo verificato la
corretta sintassi dell'istruzione... Sono abbastanza cialtrone a volte :-)

closed 2016-09-28 by plinioseniore