souliss/souliss #267: ESP 8266 Network

DD-WRT, 2016-05-29:
I made an excellent network souliss with nRF modules that works great, now the question is that I want to do network with ESP 8266, is we need a gateway or they can ride all the nodes via wifi router ??
Thanks

plinioseniore, 2016-05-29:
You can mix network as you like, so you can add as many ESP8266 as you want within the maximum limit of nodes.

Actually the only constrain is that you cannot build a node with ESP8266+nRF, you always need an AVR for that.

Are you running the latest 7.2_rc? It include some improvements for nRF nodes.

DD-WRT, 2016-05-29:
I use the new 7.2 version, one more question, if I do more nodes with the ESP and all have a different IP address that I can manage everything at once via soulis app when there can enter only one IP.

closed 2016-09-28 by plinioseniore