souliss/souliss #269: T19 example code?

wes1007, 2016-06-05:
I have looked everywhere for an example of how to set up T19/dimmable single color led strip and cannot find it anywhere.

I've looked at both T11 example code and T16 example code and cannot seem to figure it out.
Below is what I have tried

Can someone guide me with this?

`/**************************************************************************
    Souliss - Hello World
    
    This is the basic example, control one LED via a push-button or Android
    using SoulissApp (get it from Play Store).  
    
    Run this code on one of the following boards:
      - Arduino Ethernet (W5100) 
      - Arduino with Ethernet Shield (W5100)
      
    As option you can run the same code on the following, just changing the
    relevant configuration file at begin of the sketch
      - Arduino with W5200 Ethernet Shield
      - Arduino with W5500 Ethernet Shield
        
***************************************************************************/

// Configure the framework
#include "bconf/StandardArduino.h"          // Use a standard Arduino
#include "conf/ethW5100.h"                  // Ethernet through Wiznet W5100
#include "conf/Gateway.h"                   // The main node is the Gateway, we have just one node
#include "conf/Webhook.h"                   // Enable DHCP and DNS

// Include framework code and libraries
#include <SPI.h>

/*** All configuration includes should be above this line ***/ 
#include "Souliss.h"

// This identify the number of the LED logic
#define MYLEDLOGIC          0
               

void setup()
{   
    Initialize();

    // Get the IP address from DHCP
    GetIPAddress();                          
    SetAsGateway(myvNet_dhcp);       // Set this node as gateway for SoulissApp  
    
    Set_T19(MYLEDLOGIC);        // Define a simple LED light logic
    
    // We connect a pushbutton between 5V and pin2 with a pulldown resistor 
    // between pin2 and GND, the LED is connected to pin9 with a resistor to
    // limit the current amount
    pinMode(9, OUTPUT);                 // Power the LED
}

void loop()
{ 
    // Here we start to play
    EXECUTEFAST() {                     
        UPDATEFAST();   
        
        FAST_50ms() {   // We process the logic and relevant input and output every 50 milliseconds
            Logic_T19(MYLEDLOGIC);                          // Drive the LED as per command
            analogWrite(9, mOutput(MYLEDLOGIC));                // Use the pin9 to give power to the LED according to the logic
            
              
        // Here we handle here the communication with Android, commands and notification
        // are automatically assigned to MYLEDLOGIC
        FAST_GatewayComms();                                        
        
    }
} `

jkutianski, 2016-09-06:
```
/**************************************************************************
    Souliss - Hello World

    This is the basic example, control one LED dimmable strip via Android
    using SoulissApp (get it from Play Store).  

    Run this code on one of the following boards:
      - Arduino Ethernet (W5100) 
      - Arduino with Ethernet Shield (W5100)

    As option you can run the same code on the following, just changing the
    relevant configuration file at begin of the sketch
      - Arduino with W5200 Ethernet Shield
      - Arduino with W5500 Ethernet Shield

***************************************************************************/

// Configure the framework
#include "bconf/StandardArduino.h"          // Use a standard Arduino
#include "conf/ethW5100.h"                  // Ethernet through Wiznet W5100
#include "conf/Gateway.h"                   // The main node is the Gateway, we have just one node
#include "conf/Webhook.h"                   // Enable DHCP and DNS

// Include framework code and libraries
#include <SPI.h>

/*** All configuration includes should be above this line ***/ 
#include "Souliss.h"

// This identify the number of the LED logic
#define LEDCONTROL      0             
#define LEDINTENSITY    1 

void setup() {
    Initialize();

    // Get the IP address from DHCP
    GetIPAddress();                          
    SetAsGateway(myvNet_dhcp);       // Set this node as gateway for SoulissApp 

   Set_T19(LEDCONTROL);        // Define a simple LED light logic
   pinMode(9, OUTPUT);           // Power the LED
}

void loop()
{  
    EXECUTEFAST() {                     
        UPDATEFAST();   
        
        FAST_50ms() {   // We process the logic and relevant input and output every 50 milliseconds
             
            Logic_T19(LEDCONTROL);   // Drive the LED as per command
            analogWrite(15, mOutput(LEDINTENSITY)*4);   // Use the pin9 to give power to the LED according to the logic

            ProcessCommunication();
        }

        // Run communication as Gateway or Peer
        if (IsRuntimeGateway())
            FAST_GatewayComms(); 
        else 
            FAST_PeerComms();
    }

    EXECUTESLOW() {
        UPDATESLOW();
       
        // If running as Peer
        if (!IsRuntimeGateway())
            SLOW_PeerJoin();
    }
}
```

jkutianski, 2016-09-06:
```
/**************************************************************************
    Souliss - Hello World

    This is the basic example, control one LED dimmable strip via Android
    using SoulissApp (get it from Play Store).  

    Run this code on one of the following boards:
      - Arduino Ethernet (W5100) 
      - Arduino with Ethernet Shield (W5100)

    As option you can run the same code on the following, just changing the
    relevant configuration file at begin of the sketch
      - Arduino with W5200 Ethernet Shield
      - Arduino with W5500 Ethernet Shield

***************************************************************************/

// Configure the framework
#include "bconf/StandardArduino.h"          // Use a standard Arduino
#include "conf/ethW5100.h"                  // Ethernet through Wiznet W5100
#include "conf/Gateway.h"                   // The main node is the Gateway, we have just one node
#include "conf/Webhook.h"                   // Enable DHCP and DNS

// Include framework code and libraries
#include <SPI.h>

/*** All configuration includes should be above this line ***/ 
#include "Souliss.h"

// This identify the number of the LED logic
#define LEDCONTROL      0             
#define LEDINTENSITY    1 

void setup() {
    Initialize();

    // Get the IP address from DHCP
    GetIPAddress();                          
    SetAsGateway(myvNet_dhcp);       // Set this node as gateway for SoulissApp 

   Set_T19(LEDCONTROL);        // Define a simple LED light logic
   pinMode(9, OUTPUT);           // Power the LED
}

void loop() {  
    EXECUTEFAST() {                     
        UPDATEFAST();   

        FAST_50ms() {   // We process the logic and relevant input and output every 50 milliseconds

            Logic_T19(LEDCONTROL);   // Drive the LED as per command
            analogWrite(9, mOutput(LEDINTENSITY)*4);   // Use the pin9 to give power to the LED according to the logic

            ProcessCommunication();
        }

        // Run communication as Gateway or Peer
        if (IsRuntimeGateway())
            FAST_GatewayComms(); 
        else 
            FAST_PeerComms();
    }

    EXECUTESLOW() {
        UPDATESLOW();

        // If running as Peer
        if (!IsRuntimeGateway())
            SLOW_PeerJoin();
    }
}
```

plinioseniore, 2016-09-10:
Hi @jkutianski if this is not included in the default set of examples, please add it with a pull request.

Thanks.

closed 2016-09-10 by plinioseniore