souliss/souliss #272: SoulissApp not connecting - unable to ping

kstubb, 2016-08-13:
Hi Dario;

I have been trying to test a few nodes with your sample code and have e01_GarageDoor_n1.ino running on an Uno R3 with W5200 Ethernet Shield.  I can successfully ping the gateway at 192.168.70.187, to which I have it set, from any machine on my LAN, including both the tablets I have SoulissApp installed on, but the SoulissApp itself fails the ping test and will not connect to the gateway. 

User Index I have set to 87 and Node Index to 187, the last byte of the gateway IP address.

Not sure how to start debugging this as I do know my machines can reach the gateway node.

Any help would be appreciated!

Kyle

kstubb, 2016-08-13:
One tablet is a Samsung Galaxy Tab 10.1 original and is running Android 4.0.4 with SoulissApp 1.8.3.

The other is a Samsung Galaxy Tab 4 10.1 and is running Android 5.0.2 with SoulissApp 1.8.3.

Also confirmed that both nodes are communicating with each other via RS485.

I'll try adding some bindings to openhab tomorrow to see if that is connecting.

plinioseniore, 2016-08-13:
Hi Kyle,

any device with SoulissApp is a node itself, so you should not use the same
address 187 for the Gateway and the tablet.

Anyhow, SoulissApp should detect automatically the Gateway on the network
and show the IP address into the Network configuration panel. Does it?

Dario.

From Mobile.

kstubb, 2016-08-13:
The IP address of the gateway node is 192.168.70.187.

The IP address of the first tab is 192.168.70.80.
The IP address of the second tab is 192.168.70.156.

Both of the tablets can ping the gateway node (192.168.70.187) with Network Tools for Android. Both tablets show the IP and MAC address of the gateway node as being a device on the network (again, in Network Tools).

If I start SoulissApp and do not enter an IP, it does nothing other than tell me SoulissApp is not configured and I should enter the IP address of the gateway node into the local IP setting (local IP address is mandatory).  When I enter the gateway node IP (192.168.70.187), SoulissApp does not connect to the gateway and is unable to ping 192.168.70.187 with the UDP low level test tools.

Both nodes (Garage 01 and 02 respectively) are still up and running this morning (and verified the "garage door" is accepting / processing commands).  I am still able to ping the gateway via any other machine on my network.

The UDP port setting remains empty (I think I read it defaults to 230). 

Thanks for your quick response!
Kyle

plinioseniore, 2016-08-14:
Looks like that the network support of your tablet doesn't get a proper support in SoulissApp, as it should detect automatically the IP address and have the 230 port set as default. If is empty, please insert manually 230.

Then in the same configuration panel you should have also Node Index and User Index parameters, those are supposed to be generated automatically based on the MAC address of your phone, if so is enough that you ensure that you tablets has a different couple for those two values and no manual action is required.

Let me know,
Dario.

kstubb, 2016-08-14:
Hi Dario;

I have tried all the things you suggested and have filled in all the settings manually, but am unable to connect with SoulissApp.

Last night I also setup some bindings in OpenHAB yesterday and could not get the gateway to give me any heath readings or any indication that it is connecting, other than the link light flashing on the Ethernet shield.

So, this morning, I decided I'd go right to basics with this and load up the e01_HelloWorld.ino sketch and see if assigning an IP via DHCP would have any effect.  Unfortunately, I couldn't even obtain an IP address with this sketch.  I put some breakpoints in and it blocks at GetIPAddress();

I'm starting to think this is related to the W5200 driver as I can load up multiple sketches from the Seeed Studio W5200 library, obtain an IP address and grab a webpage from Google.  (I'd give you a link, but Seeed Studio's wiki seems to be giving a 500 error right now.)

I am still able to PING the device (with network tools, not with SoulissApp) from any computer on my LAN when I manually assign an IP address to the gateway node.

I did have a look through my DHCP logs and did see that the gateway actually did go through the DISCOVER / OFFER / PACK process to obtain an IP once.  I did change the MAC address to see if I could get another DISCOVER, but nothing yet.  I'll try clearing the DHCP leases later and try again.

I am using a WizNet 5200 shield on an Arduino UNO R3 and have changed the include to conf/ethW5200.h.

I have looked at all the config files and changed default values where appropriate (such as base IP, gateway, WizNet chip, etc.).

I've tried to read through everything I can on the Wiki, but I could have been tired, so I'll browse through it again to see if I have missed anything.

I am using Arduino IDE 1.6.7 and did find the tip about adding the blank file to overcome compile errors and had been using the stable branch that way.  This morning I removed the stable branch of the library and installed the development branch to see if that had any effect.  Same results.

I'll see what else I can find and let you know.

Thanks;
Kyle

plinioseniore, 2016-08-14:
Hi Kyle,

do not, your are in the wrong direction. If you are able to perform an IP PING that means that your skech is initializing the W5200 properly with a working network configuration. By the way, is the W5200 that handle the IP stack as well as the PING, so if you are pinging this means that you are just communicating with the W5200 as this doesn't involve the MCU as well as your sketch (out of the init setup of the network parameters).

You should instead go for a [data sniffing](https://github.com/souliss/souliss/wiki/Data%20Sniffing) if you want to understand what's happening in your case.

Dario.

kstubb, 2016-08-15:
Hi Dario;

Thank you for the suggestion of data sniffing.  I put my laptop between the gateway node and my network and watched things with Wireshark.

Unfortunately, I only see traffic going to the gateway and nothing coming back from it for requests to UDP 230.

Well, in most of the logged lines to the gateway, the gateway is the destination device.  I see UDP requests coming from 192.168.70.14, which is the Pi running OpenHAB, but nothing back to the Pi.  I see requests coming in from both 192.168.70.56 & 192.168.70.80, which are my tablets, but nothing back to them.  If I use network tools from either tablet and ping the gateway, I can see ICMP requests to and from the gateway.  I can PING the gateway from any other computer on my LAN and communication is seen back and forth for the ICMP requests.

(openhab) 192.168.70.14    192.168.70.187    UDP    60    Source port: 56658    Destination port: 230 
(tablet 1)   192.168.70.80    192.168.70.187    UDP    60    Source port: 23000    Destination port: 230  (I'll get 10 or 15 of these in a row -- the lines are green(ish) in Wireshark)

I don't see any replies at all from these requests.

Here's where it gets weird... every once in a while (seemingly random) I will get a response from the gateway but it's in the wrong protocol.  What I mean is, you can see the UDP request to port 230 to the gateway, but the gateway replies via ICMP and receives a "70 Destination unreachable (Port unreachable)" error (the line is black in Wireshark).  I have also noticed the port unreachable message on port 53 when network tools sends out DNS queries.  The thing is, I do see it responding to PING, DNS and DHCP, but no response to the UDP requests.

(tablet 2)   192.168.70.56    192.168.70.187    DNS    87   Standard query 0xa0c3  PTR 187.70.168.192.in-addr.arpa   
(gateway)  192.168.70.187  192.168.70.56     ICMP   70   Destination unreachable (Port Unreachable)  which is port 53  

Actually, when I look through the data for these lines, it does show UDP (17) as the protocol on port 53 for both the request and the reply, so port 53 is not being reached.

I have made sure the firewall is off on the Windows laptop.  The rest of my network is Linux with the proper ports open to allow traffic (although the local IPs are trusted and will bypass the firewall rules anyway... I just wanted to make sure).  There is no firewall on the access point.

Well, I have to go to work this morning, so I will do a little more experimenting this evening.  Any suggestions are appreciated!

Kyle

plinioseniore, 2016-08-15:
Go for [vNet and MaCaco debug](https://github.com/souliss/souliss/wiki/TroubleShootSouliss) via serial port, then attach your sketches with the logs from Wireshark and serial.

kstubb, 2016-08-17:
Hi Dario;

Sorry, I haven't had much time to do any testing... but I will!

So far the only output I have been able to capture from the vNet and MaCaco debug are upon initial power up and are as follows;

> (vNet)<USERMODE><0x0|0x0|0x0|0x0|0x0|0x>
(vNet)<SETADDR><|0xBB>
(MAC)<0,0,0,0,0,0,>
(IP)<0,0,0,0,>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x4E><|0xB|0x17|0x4E|0xBB>

I don't know a lot about these debug lines, but it does appear that neither the MAC or IP are set in software and data seems to be missing (before the first pipe on each line).

The sketch is the standard e02_GarageDoor_n1.ino in the latest code.  The only things I changed were the IP (192.187.70.187) and gateway (192.168.70.1) and the conf/ethW5100.h to conf/ethW5200.h.  The reason I am using this sketch is because most of the rest use DHCP which will not even give me an IP.

The logs from Wireshark remain the same as I stated above.  No replies from 192.168.70.187 (the gateway) other than PING, which you already stated is handled by the W5200 itself.

I looked at the local electronics shop to see if they had any other Ethernet shields (ie. W5100), but are getting some WiFi shields in later this week. I might grab one to see if the results are the same.

Regards;
Kyle

plinioseniore, 2016-08-17:
Paste your sketch. You are not initializing the IP stack.

Dario.

kstubb, 2016-08-17:
Sorry, I tried blockquote, code and everything else I could think of to get this to format correctly, nor could I find a way to turn markdown off.

This is the standard e02_GarageDoor_n1.ino example with the IP changed and Serial.begin(9600); added for debug.

/**************************************************************************
    Souliss - Garage Door
    
    Control a garage door using two Ethernet boards: one device act on the 
    relays that drive the motor and get the limit switches, the other has one 
    push-button for opening and closing the door. The door can be controlled
    also via Android (download SoulissApp from Play Store).
    
    Ensure to use limit switches to protect the motor once the door is completely
    closed or open, if limit switches are not used the motor should be self-protected.
        
    Run this code on one of the following boards:
      - Arduino Ethernet (W5100) 
      - Arduino with Ethernet Shield (W5100)
      
    As option you can run the same code on the following, just changing the
    relevant configuration file at begin of the sketch
      - Arduino with ENC28J60 Ethernet Shield
      - Arduino with W5200 Ethernet Shield
      - Arduino with W5500 Ethernet Shield
            
***************************************************************************/

// Let the IDE point to the Souliss framework
#include "SoulissFramework.h"

// Configure the framework
#include "bconf/StandardArduino.h"          // Use a standard Arduino
#include "conf/ethW5200.h"                  // Ethernet through Wiznet W5100
#include "conf/Gateway.h"                   // The main node is the Gateway

// Include framework code and libraries
#include <SPI.h>

/*** All configuration includes should be above this line ***/ 
#include "Souliss.h"

// Define the network configuration according to your router settings
uint8_t ip_address[4]  = {192, 168, 70, 187};
uint8_t subnet_mask[4] = {255, 255, 255, 0};
uint8_t ip_gateway[4]  = {192, 168, 70, 1};
#define Gateway_address 77
#define Peer_address    78
#define myvNet_address  ip_address[3]       // The last byte of the IP address (77) is also the vNet address
#define myvNet_subnet   0xFF00
#define myvNet_supern   Gateway_address

#define GARAGEDOOR_NODE1            0                       
#define GARAGEDOOR_NODE2            0   
#define GARAGELIGHT_NODE2           1

#define INPUT_PIN	2 // Use Pin2 as ON/OFF command

void setup()
{   
    Serial.begin(9600);
    Initialize();
    
    // Set network parameters
    Souliss_SetIPAddress(ip_address, subnet_mask, ip_gateway);
    SetAsGateway(myvNet_address);                                   // Set this node as gateway for SoulissApp  
    
    // This node as gateway will get data from the Peer
    SetAsPeerNode(Peer_address, 1); 
    
    // Define inputs, outputs pins and pulldown
    pinMode(INPUT_PIN, INPUT);  // Hardware pulldown required
    
}

void loop()
{   
    // Here we start to play
    EXECUTEFAST() {                     
        UPDATEFAST();   

        // Read every 510ms the input state and send it to the other board  
        FAST_510ms() {
            
            // Use Pin2 as ON/OFF command
            if(DigIn(INPUT_PIN, Souliss_T2n_ToggleCmd, GARAGEDOOR_NODE1))
            {
                // Send data
                Send(Peer_address, GARAGEDOOR_NODE2, mInput(GARAGEDOOR_NODE1));
                ResetInput(GARAGEDOOR_NODE1);
            }
        } 
        
        // Here we handle here the communication with Android, commands and notification
        // are automatically assigned to MYLEDLOGIC
        FAST_GatewayComms();        
                        
    }
}

kstubb, 2016-08-17:
This is the e01_HelloWorld.ino sketch that I tried on Arduino IDE 1.6.5-r5 on Windows 7.

/**************************************************************************
    Souliss - Hello World
    
    This is the basic example, control one LED via a push-button or Android
    using SoulissApp (get it from Play Store).  
    
    Run this code on one of the following boards:
      - Arduino Ethernet (W5100) 
      - Arduino with Ethernet Shield (W5100)
      
    As option you can run the same code on the following, just changing the
    relevant configuration file at begin of the sketch
      - Arduino with W5200 Ethernet Shield
      - Arduino with W5500 Ethernet Shield
        
***************************************************************************/

// Configure the framework
#include "bconf/StandardArduino.h"          // Use a standard Arduino
#include "conf/ethW5200.h"                  // Ethernet through Wiznet W5200
#include "conf/Gateway.h"                   // The main node is the Gateway, we have just one node
#include "conf/Webhook.h"                   // Enable DHCP and DNS

// Include framework code and libraries
#include <SPI.h> (this is include SPI.h... formatting is removing it).

/*** All configuration includes should be above this line ***/ 
#include "Souliss.h"

// This identify the number of the LED logic
#define MYLEDLOGIC          0               

void setup()
{   
    Initialize();

    // Get the IP address from DHCP
    GetIPAddress();                          
    SetAsGateway(myvNet_dhcp);       // Set this node as gateway for SoulissApp  
    
    Set_SimpleLight(MYLEDLOGIC);        // Define a simple LED light logic
    
    // We connect a pushbutton between 5V and pin2 with a pulldown resistor 
    // between pin2 and GND, the LED is connected to pin9 with a resistor to
    // limit the current amount
    pinMode(2, INPUT);                  // Hardware pulldown required
    pinMode(9, OUTPUT);                 // Power the LED
}

void loop()
{ 
    // Here we start to play
    EXECUTEFAST() {                     
        UPDATEFAST();   
        
        FAST_50ms() {   // We process the logic and relevant input and output every 50 milliseconds
            DigIn(2, Souliss_T1n_ToggleCmd, MYLEDLOGIC);            // Use the pin2 as ON/OFF toggle command
            Logic_SimpleLight(MYLEDLOGIC);                          // Drive the LED as per command
            DigOut(9, Souliss_T1n_Coil, MYLEDLOGIC);                // Use the pin9 to give power to the LED according to the logic
        } 
              
        // Here we handle here the communication with Android, commands and notification
        // are automatically assigned to MYLEDLOGIC
        FAST_GatewayComms();                                        
        
    }
}

plinioseniore, 2016-08-18:
Which shield are you using?

kstubb, 2016-08-18:
W5200 --> https://www.creatroninc.com/product/w5200-ethernet-shield/

plinioseniore, 2016-08-18:
Quite strange, looking at the W5200 shield and the relevant libraries and example looks that CS is pin 10 as we use in Souliss (and as is used in most of the Ethernet Shield for Arduino).

Your problem is on the communication between the AVR MCU and the W5200, likely the AVR is able to send data but not to receive data, I assume this because if you can PING this means that the AVR has set the IP address into the W5200.

As you can see [here](https://github.com/souliss/souliss/blob/friariello/frame/vNet/drivers/mcu_avr/ethW5x00/vNetDriver_eth.cpp#L138) the printout of MAC and IP address is not from the AVR RAM but is read directly from the W5200.

```C
// Include debug functionalities, if required
#if(VNET_DEBUG)
uint8_t addrval[6];

// Print MAC address 
W5x00.getMACAddress(addrval);
VNET_LOG("(MAC)<");
for(U8 i=0; i<6; i++)
{
	VNET_LOG(addrval[i],HEX);
	VNET_LOG(",");
}
VNET_LOG(">\r\n");

// Print IP address 
W5x00.getIPAddress(addrval);
VNET_LOG("(IP)<");
for(U8 i=0; i<4; i++)
{
	VNET_LOG(addrval[i],HEX);
	VNET_LOG(",");
}

VNET_LOG(">\r\n");
#endif	
```

We actually know the problem, but I'm quite far from the solution. I assume that you got this shield running the base example provided by seller successfully, is it?

kstubb, 2016-08-18:
Hi Dario;

Yes, I am able to get it to work with the Seeed Studio W5200 Ethernet
library -> https://github.com/Seeed-Studio/Ethernet_Shield_W5200

I have tried a few of the examples from that library and am able to get
results back from a Google search the example uses.

I'm on the road right now and don't remember which ones, but I don't recall
having any issues with any that I tried.

Kyle

kstubb, 2016-08-19:
It's responding!

> (vNet)<USERMODE><0x0|0x0|0x0|0x0|0x0|0x>
(vNet)<SETADDR><|0xBB>
(MAC)<1A,A6,49,6B,0,BD,>
(IP)<C0,A8,46,BB,>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x4E><|0xB|0x17|0x4E|0xBB>
(vNet)<USERMODE><0xFBB|0x0|0x0|0x0|0x0|0x>
(vNet)<IN>(0x50)<|0xB|0x17|0xBB|0xFBB|0x8|0xF|0x0|0x0|0x0>
(MaCaco)<IN><|0x8|0xF|0x0|0x0|0x0|0x>
(MaCaco)<OUT><|0x18|0xF|0x0|0x0|0x0|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0xFBB><|0xB|0x17|0xFBB|0xBB>
(vNet)<USERMODE><0xFBB|0x402|0x0|0x0|0x0|0x>
(vNet)<IN>(0xE)<|0xB|0x17|0xBB|0x402|0x21|0x0|0x0|0x0|0x2>
(MaCaco)<IN><|0x21|0x0|0x0|0x0|0x2|0x>
(MaCaco)<OUT><|0x31|0x0|0x0|0x0|0x18|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x402><|0x23|0x17|0x402|0xBB>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x4E><|0xB|0x17|0x4E|0xBB>
(vNet)<USERMODE><0xFBB|0x402|0xF50|0x0|0x0|0x>
(vNet)<IN>(0x50)<|0xB|0x17|0xBB|0xF50|0x8|0xF|0x0|0x0|0x0>
(MaCaco)<IN><|0x8|0xF|0x0|0x0|0x0|0x>
(MaCaco)<OUT><|0x18|0xF|0x0|0x0|0x0|0x>

After you asking if it worked with the demo code, it got me to thinking.... yes, it does work... so what's different?

>   // On the Ethernet Shield, CS is pin 4. It's set as an output by default.
  // Note that even if it's not used as the CS pin, the hardware SS pin 
  // (10 on most Arduino boards, 53 on the Mega) must be left as an output 
  // or the SD library functions will not work.  
  pinMode(SDCARD_CS,OUTPUT);

I added the same to the setup() of the e02_GarageDoor_n1.ino sketch and it started to work!  
I never even thought about the SD card as I didn't really care about using it.

> pinMode(4,OUTPUT);

What it's still not doing is connecting to the SoulissApp.  :-(

The data you see is coming from my openHAB bindings.

We're half way there!

Kyle

kstubb, 2016-08-19:
It seems that simply adding

> pinMode(4, OUTPUT);

to a sketch where you manually assign a static IP is enough.

However, a DHCP sketch (e01_HelloWorld.ino) will not acquire an IP without the pin set high.

>pinMode(4, OUTOUT);
digitalWrite(4, HIGH);

e01_HelloWorld.ino now gets an IP, but still doesn't connect with SoulissApp.

plinioseniore, 2016-08-19:
Have you the SD card inserted?

Anyhow, could you please post the log with SoulissApp only?

kstubb, 2016-08-19:
I did have the SD card inserted last night as the demo code for this shield required it.  I just forgot about it and left it in.  This morning I removed it and tried some more.  

Here is the current debug log.  The IP has been changed as it's getting it via DHCP (192.168.70.134), so I only have that IP in the SoulissApp (not openHAB), so the logs should only be from SoulissApp.

One further note, I created a new test profile in SoulissApp for this test and when I put the IP address in it comes up with 192.168.70.134 (WIZnet6B0002.mydomain.com).  It did not show the WIZnet6B0002 before.  It always simply showed the IP address.  I've got 14 as my User Index and 80 as my Node Index (the tablet IP is 192.168.70.80), so it does look like SoulissApp is contacting the gateway.  I have also noticed that the "Synch" icon does change and goes to "Offline" after a short period.

The test UDP PING comes back with "Command timeout expired, no reply received", but I am still able to PING the gateway from Network Tools in Android (on the same tablet), or from any other machine on my LAN.

> (Net)<USERMODE><0x0|0x0|0x0|0x0|0x0|0x>
(vNet)<USERMODE><0x0|0x0|0x0|0x0|0x0|0x>
(vNet)<SETADDR><|0x86>
(MAC)<1A,A6,49,6B,0,88,>
(IP)<C0,A8,46,86,>  - markdown is removing the ip C0,A8,46,86
(vNet)<USERMODE><0x2C50|0x0|0x0|0x0|0x0|0x>
(vNet)<IN>(0x38)<|0xB|0x17|0x86|0x2C50|0x8|0xF|0x0|0x0|0x0>
(MaCaco)<IN><|0x8|0xF|0x0|0x0|0x0|0x>
(MaCaco)<OUT><|0x18|0xF|0x0|0x0|0x0|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x2C50><|0xB|0x17|0x2C50|0x86>
(vNet)<IN>(0x38)<|0xB|0x17|0x86|0x2C50|0x8|0xF|0x0|0x0|0x0>
(MaCaco)<IN><|0x8|0xF|0x0|0x0|0x0|0x>
(MaCaco)<OUT><|0x18|0xF|0x0|0x0|0x0|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x2C50><|0xB|0x17|0x2C50|0x86>
(vNet)<IN>(0x38)<|0xB|0x17|0x86|0x2C50|0x8|0xF|0x0|0x0|0x0>
(MaCaco)<IN><|0x8|0xF|0x0|0x0|0x0|0x>
(MaCaco)<OUT><|0x18|0xF|0x0|0x0|0x0|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x2C50><|0xB|0x17|0x2C50|0x86>
(vNet)<IN>(0x38)<|0xB|0x17|0x86|0x2C50|0x8|0xF|0x0|0x0|0x0>
(MaCaco)<IN><|0x8|0xF|0x0|0x0|0x0|0x>
(MaCaco)<OUT><|0x18|0xF|0x0|0x0|0x0|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x2C50><|0xB|0x17|0x2C50|0x86>
(vNet)<IN>(0x38)<|0xB|0x17|0x86|0x2C50|0x8|0xF|0x0|0x0|0x0>
(MaCaco)<IN><|0x8|0xF|0x0|0x0|0x0|0x>
(MaCaco)<OUT><|0x18|0xF|0x0|0x0|0x0|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x2C50><|0xB|0x17|0x2C50|0x86>
(vNet)<IN>(0x38)<|0xB|0x17|0x86|0x2C50|0x8|0xF|0x0|0x0|0x0>
(MaCaco)<IN><|0x8|0xF|0x0|0x0|0x0|0x>
(MaCaco)<OUT><|0x18|0xF|0x0|0x0|0x0|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x2C50><|0xB|0x17|0x2C50|0x86>
(vNet)<IN>(0x38)<|0xB|0x17|0x86|0x2C50|0x8|0xF|0x0|0x0|0x0>
(MaCaco)<IN><|0x8|0xF|0x0|0x0|0x0|0x>
(MaCaco)<OUT><|0x18|0xF|0x0|0x0|0x0|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x2C50><|0xB|0x17|0x2C50|0x86>
(vNet)<IN>(0x38)<|0xB|0x17|0x86|0x2C50|0x8|0xF|0x0|0x0|0x0>
(MaCaco)<IN><|0x8|0xF|0x0|0x0|0x0|0x>
(MaCaco)<OUT><|0x18|0xF|0x0|0x0|0x0|0x>
(vNet)<OUTPATH><|0x1>
(vNet)<MEDIA><|0x1>
(vNet)<OUT><DADDR><|0x2C50><|0xB|0x17|0x2C50|0x86>

kstubb, 2016-08-19:
I just did another test and am attaching the file, in case you cannot read the one above (markdown is removing information).

This one, I started SoulissApp, created a new profile with the IP of 192.168.70.134.  Then I modified the profile with UDP port 230, User Index 14, Node Index 80.  "Synch" now shows red, where it used to show yellow and go red when "offline".

This is the log from that transaction.

[vnet_debug_01.txt](https://github.com/souliss/souliss/files/427015/vnet_debug_01.txt)

plinioseniore, 2016-08-20:
The 0x08 is the ping request and 0x18 is the answer, we know that your node
receive and process the data but likely data reply doesn't go on the net.

A Wireshark capture will help now. Generally is not that trivial to get
Souliss running, don't give up.

Dario.

From Mobile.

kstubb, 2016-08-20:
No worries, I won't give up!

The first part was entirely my fault by not initializing the SD card.

I'm on the road a bit today but will work on it this evening.

Regards;
Kyle

kstubb, 2016-08-21:
HI Dario;

Tonight I hook everything up to capture some Wireshark logs.  I have my eth and wlan ports bridged on my laptop, so the gateway was pulling the laptops IP from the DHCP server (router) when using the HelloWorld sketch.

So, I load up another of the examples I've been using (e02_GarageDoor_n1.ino) that use static IP assignment and use the IP of 192.168.70.187 that I've been using in that sketch.  I immediately notice my openHAB server trying to contact the gateway, so I decided to change the IP to 192.168.70.199 so that I can isolate the gateway and the tablet communications, and SoulissApp connected to the gateway.

I test things out a little on SoulissApp and it seems to be connected and responsive.

I changed the IP of the gateway back to 192.168.70.187 and both openHAB and SoulissApp are connecting to the gateway.  This is still hooked up through the laptop and Wireshark shows proper communication between the gateway and both the tablet / openHAB server.

I'm confused as to why it is working now.

I'll do some more testing and let you know if I encounter any further issues.

Thank you for your patience in helping me sort this out!

kstubb, 2016-08-22:
Hi Dario;

Well, I'm back to frustrated.  

I didn't get any time yesterday to work on it, so I sit down this morning and nothing works.  I jsut can't understand it.

So, I start digging around again and find that when I use the e01_HelloWorld.ino sketch, which uses DHCP to assign an IP, I get the IP address of the laptop (192.169.70.97) if I am on the eth bridged port (I don't understand this either) and it does not show up on my router as a listed device.

If I plug it into my switch, I get the IP address of 192.168.70.134, but I get it twice with 2 different MAC addresses (see attached image).  One shows it is attached to the br0 port, the other isn't attached to anything, but that is the one that has the DHCP lease.

The next screen shot shows that I tried to PING the original IP address of 192.168.70.187, which fails.  Then I PING the DHCP assigned IP of 192.168.70.134 and get a response.  The device list on my router now only shows 1 IP for the WizNET, but it doesn't show as connected to the br0 port.  You can also see the IP / MAC assigned to the gateway in the e01_HelloWorld.ino sketch.

Gotta go to work.... will dig a little more later.

Kyle
[image]
[image]

plinioseniore, 2016-08-22:
Hi Kyle,

I cannot see any problem, you are just getting a different IP address while in DHCP, so changing the IP address in SoulissApp should let you start the connection with the Souliss Gateway node on the .134 address.

The reason you see two MAC addresses is just due to the persistence of previous entries in your router MAC table, but it will disappear after few time. Have a read [here](https://github.com/souliss/souliss/wiki/Addressing#define-a-mac-address), you are using (as default) the **AUTO_MAC** option, so the node start with a default MAC address and later change it to a local-unique one once it got the IP address from the DHCP server.

For the bridge case, looks suspicious, as you should have both the interfaces (eth and wifi) on the same local network and the router should be the only DHCP server.

kstubb, 2016-08-23:
HI Dario;

I understand what you are saying, but that is the issue... I cannot connect with SoulissApp to .134 (DHCP).

What I did notice in the pictures above is what you are saying about the two IPs makes sense, but have a closer look.... the first image shows 2 MAC addresses with .134.  You'll notice that the top one is assigned to the bridge (br0) and has a MAC of :88 which was assigned by Souliss.  Note is DOES NOT have a DHCP lease.  The second entry has a MAC address of :02, which is the MAC address the WizNET has.  Note that it IS NOT connected to the bridge but it DOES have a DHCP lease (and a hostname).

In the second image, you will notice that the MAC address that persists is :02, the ORIGINAL MAC address of the WizNET chip, NOT the assigned MAC address by Souliss.  You can see the serial monitor window at the bottom which shows Souliss assigning the MAC :88.

So, Souliss thinks it has a MAC of 1A:A6:49:6B:00:88 and the router thinks it has a MAC address of 1A:A6:49:6B:00:02.  When I PING it, which MAC is actually responding?

As for the bridge... I am talking about my Windows laptop so I can capture traffic with Wireshark.  I setup the LAN and WLAN ports as a bridge, which I assume would simply act as a pass-through.  I have the WLAN connected to my router and the arduino gateway connected to the LAN port.  The router is the only DHCP server and assigns 192.168.70.97 to the laptop WLAN connection.  When I use a sketch with DHCP on the arduino, it gets 192.168.70.97 as it's IP address.  This is an obvious conflict, but I don't know why it happens, or how to solve it.  So, I can't capture traffic on any sketch that uses DHCP.  I'll have to do some more Googling / reading on this.

I have been able to get it working with a static IP (.187) and setting the MAC address to the original MAC of the WizNET (:02).  Wireshark shows good logs (.187 is also in my openHAB bindings) of communications to and from .187 to .56 (tablet - SoulissApp) and .14 (openHab).  However, the router shows the MAC address of the WLAN port of the laptop, so there is obviously some NAT going on there.

If I plug it directly into my switch, I now see the MAC address of the WizNET chip show up in my device list of the router and am still able to connect via SoulissApp and openHAB.

I'll keep an eye on it and keep testing!

Kyle

plinioseniore, 2016-08-23:
Just ping it with -t option, if your PC doesn't get reply it will ask again the MAC address using the ARP protocol. So on wireshark (even if you are not on bridge) you will see "Who has 192.168..." request, of course you will not see the reply if you are not bridging.

If is just a mismatch on the MAC address while using DHCP, you can easily fix it with a manual assignment as reported in the wiki page that I've linked before. Consider that 1A MAC addresses are local ones, so you are free to assign by your self.

kstubb, 2016-08-23:
I did set the MAC address last night to 1A:A6:49:6B:00:02 and it does not fix the issue.  

I went to be last night and left everything running... openHAB and SoulissApp both connected.  I get up this morning and only openHAB is still connected.

The only thing I can do to get SoulissApp is to reboot the router.  I don't really get it.  My tablet can still PING (ICMP) the gateway, as can the rest of my computers.  Network Tools on the tablet shows the gateway connected (I'm assuming Network Tools is using the -t option as I can see the "Who has" messages in Wireshark.

Well, I've never had any real problems with my network to speak of, but it does seem to be pointing towards my router.  I am running a router with Tomato firmware (ShibbyUSB), so I will try to load up some different firmware (maybe Toastman) and give it another go.

Of to work I go. Have a good day!  ;-)

plinioseniore, 2016-08-23:
Are your MAC addresses unique?

From Mobile.

kstubb, 2016-08-23:
I just had a look through my device list and they are all unique.

The only local MAC (1A:) is the arduino gateway.

plinioseniore, 2016-08-23:
I mean, your Arduinos has different MAC. Is it?

From Mobile.

kstubb, 2016-08-23:
I'm actually only running 1 Arduino at this time.  I'm trying to keep things very basic until I can get stable operation on the gateway.

closed 2017-08-16 by plinioseniore