souliss/souliss #273: Impossible to display temperature and humidity on soulissapp

tjsavioli, 2016-08-20:
Hi,
my name is Valerio i'm doing my first steps with souliss.
I have an arduino uno board with ethernet shield, two pushbuttons, two leds and an DHT22 connected on pin2. I can manage the tw leds with buttons and phone but i can't see temperature and humidity.
This is the code can someone helpme please? Thanks

[souliss_test2.zip](https://github.com/souliss/souliss/files/428064/souliss_test2.zip)

ribico, 2016-08-20:
Hello Valerio,
I am from mobile and I cannot read your code.
Are you able to read the DHT22 from any other example code ? This will
ensure it is properly wired and working as expected regardless souliss.

tjsavioli, 2016-08-20:
Thank you for answer.I tried different example but nothing. From serial monitor i can see something, nan temperature and strange floating values of humidity. But from my phone i can't see the typicals t52 and t53. I have two sensors, this night when childrends sleep, i wiil try with other sensor. ByeInviato dal mio dispositivo Huawei-------- Messaggio originale --------Oggetto: Re: [souliss/souliss] Impossible to display temperature and humidity on soulissapp (#273)Da: ribico A: souliss/souliss CC: tjsavioli ,Author Hello Valerio,
I am from mobile and I cannot read your code.
Are you able to read the DHT22 from any other example code ? This will
ensure it is properly wired and working as expected regardless souliss.

tjsavioli, 2016-08-20:
Thanks Ribico
i tried your dht tester and now i'm sure the sensor works.
I modified the  code according to your test and now i can read from serial corret values from dht22.
The problem remain on soulissapp, i can't see the typicals. Tried to delete nodes and devices but when i ask a new configuration nothing appens. I have comunication with the gateway but soulissapp tell me "DB empty"  ribuilt model?

tjsavioli, 2016-08-20:
Sorry forgot the code:

// Configure the framework
#include "bconf/StandardArduino.h"          // Use a standard Arduino
#include "conf/ethW5100.h"                  // Ethernet through Wiznet W5100
#include "conf/Gateway.h"                   // The main node is the Gateway, we have just one node
#include "conf/Webhook.h"                   // Enable DHCP and DNS

// Include framework code and libraries
#include "Souliss.h"
#include "Typicals.h"
#include "SPI.h"
#include <DHT.h>

/*** All configuration includes should be above this line ***/

// This identify the number of the LED logic
#define MYLEDLOGIC_1          0     //slot0             
#define MYLEDLOGIC_2          1     //slot1

// Sensor management
//#define DHT_id1           1     // Identify the sensor, in case of more than one used on the same board
#define DHTPIN            2     //data pin for dht sensor
#define DHTTYPE           DHT22
#define TEMPERATURE       2     // This is the memory slot used for the execution of the logic in network_address1
#define HUMIDITY          4     // This is the memory slot used for the execution of the logic
#define DEADBAND       0.01     // Deadband value 1%

// Setup the DHT sensor
DHT dht(DHTPIN, DHTTYPE);        

void setup()
{
  Initialize();
  Serial.begin(9600);      // open the serial port at 9600 bps:
  
   // Define the network configuration according to your router settings
  uint8_t ip_address[4]  = {192, 168, 1, 77};
  uint8_t subnet_mask[4] = {255, 255, 255, 0};
  uint8_t ip_gateway[4]  = {192, 168, 1, 1};

  Souliss_SetIPAddress(ip_address, subnet_mask, ip_gateway);  // Set network parameters
  SetAsGateway(myvNet_dhcp);                                  // Set this node as gateway for SoulissApp

  //Set typicals to use
  Set_SimpleLight(MYLEDLOGIC_1);
  Set_SimpleLight(MYLEDLOGIC_2);
  Set_Temperature(TEMPERATURE);
  Set_Humidity(HUMIDITY);
  dht.begin();

  pinMode(4, INPUT);                  // Hardware pulldown required
  pinMode(3, INPUT);                  // Hardware pulldown required
  pinMode(9, OUTPUT);                 // Power the LED
  pinMode(8, OUTPUT);                 // Power the LED

}

void loop()
{
  // Here we start to play
  EXECUTEFAST() {
    UPDATEFAST();

    FAST_50ms() {   // We process the logic and relevant input and output every 50 milliseconds
      DigIn(3, Souliss_T1n_ToggleCmd, MYLEDLOGIC_1);            // Use the pin2 as ON/OFF toggle command
      Logic_SimpleLight(MYLEDLOGIC_1);                          // Drive the LED as per command
      DigOut(8, Souliss_T1n_Coil, MYLEDLOGIC_1);                // Use the pin9 to give power to the LED according to the logic
      DigIn(4, Souliss_T1n_ToggleCmd, MYLEDLOGIC_2);            // Use the pin3 as ON/OFF toggle command
      Logic_SimpleLight(MYLEDLOGIC_2);                          // Drive the LED as per command
      DigOut(9, Souliss_T1n_Coil, MYLEDLOGIC_2);               // Use the pin10 to give power to the LED according to the logic
    }

    FAST_2110ms() {
      Souliss_Logic_T52(memory_map, TEMPERATURE, DEADBAND, &data_changed);
      Souliss_Logic_T53(memory_map, HUMIDITY, DEADBAND, &data_changed);
      float read_temperature = dht.readTemperature();
      ImportAnalog(TEMPERATURE, &read_temperature);
      float read_humidity = dht.readHumidity();
      ImportAnalog(HUMIDITY, &read_humidity);

      Serial.print("Humidity: "); 
      Serial.print(read_humidity);
      Serial.print(" %\t");
      Serial.print("Temperature: "); 
      Serial.print(read_temperature);
      Serial.print(" *C ");
    }
    
     FAST_GatewayComms();
  }
}

ribico, 2016-08-21:
Yes.. Try rebuild.

tjsavioli, 2016-08-21:
Impossible to rebuilt, nothig happensTest UDP ok i can manage the two leds on bard.Trided to change index user andindex node without result.Bye

tjsavioli, 2016-08-21:
Solved the problem .... 
I enabled the DHCP in the sketch then I deleted the static addressin the  app .
At this point I requested the definition of the devices in the UDP test, then in options , I deleted the devices and recreated the node configuration .
In this way the 4 typicals have appeared correctly .
At the end I put the static IP address in the sketch , and then in the app.
Now everything is working correctly on my smartphone .
Sorry is this a normal procedure ?
Tomorrow I'll try to install the app on the tablet to see if works.
Bye

ribico, 2016-08-22:
Absolutely not.
You can normally set static IP without your nice procedure.
I have about 12 DHT22 comnected to my home souliss network since 2 years..

All boards use static IP address.

BR

closed 2017-08-16 by plinioseniore