souliss/souliss #274: Esp8266 peer changing status button app random

lucasmaziero, 2016-08-20:
Hello,
I'm using a ESP8266 with gateway , but when I put another peer ESP8266 it is changing the status button app, when is clicked it works well, this error only occurs in the peer node, the gateway it works ok, someone knows what can be??

I am using the example "souliss \ examples \ WiFi \ e01_Hello_ESP8266" and "souliss \ examples \ WiFi \ e01_Peer_ESP8266"

sorry my English :)

plinioseniore, 2016-08-20:
Likely you need a pull-down on your inputs, as you are getting EMI on the
input pin.

Dario.

From Mobile.

lucasmaziero, 2016-08-20:
Hello,
This is my code of ESP gateway, all ok.

```ruby
// Let the IDE point to the Souliss framework
#include "SoulissFramework.h"

// Configure the framework
#include "bconf/MCU_ESP8266.h"              // Load the code directly on the ESP8266
#include "conf/Gateway.h"                   // The main node is the Gateway, we have just one node
#include "conf/IPBroadcast.h"

// **** Define the WiFi name and password ****
#define WIFICONF_INSKETCH
#define WiFi_SSID               "xxxxx"
#define WiFi_Password           "xxxxx"

// Include framework code and libraries
#include <ESP8266WiFi.h>
#include <EEPROM.h>

/*** All configuration includes should be above this line ***/
#include "Souliss.h"

// This identify the number of the LED logic
#define LIGHT1_SLOT0 0
#define LIGHT2_SLOT1 1

#define ANALOG_VALUE_SLOT2 2 // Analog values use two slot, in this case 2 and 3
int cont = 0;

// **** Define here the right pin for your ESP module ****
#define	PIN_LIGHT1 D4
#define PIN_SWITCH1 D3

#define  PIN_LIGHT2 D5

void setup()
{
  Serial.begin(115200);

  Serial.println("\n");
  Serial.print("Connecting to ");
  Serial.println(WiFi_SSID);

  WiFi.begin(WiFi_SSID, WiFi_Password);

  //WiFi.disconnect();
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  Initialize();

  // Connect to the WiFi network and get an address from DHCP
  GetIPAddress();
  SetAsGateway(myvNet_dhcp);       // Set this node as gateway for SoulissApp

  // This is the vNet address for this node, used to communicate with other
  // nodes in your Souliss network
  SetAddress(0xAB01, 0xFF00, 0xAB01);

  SetAsPeerNode(0xAB02, 1);

  Set_SimpleLight(LIGHT1_SLOT0);        // Define a simple LED light logic
  Set_SimpleLight(LIGHT2_SLOT1);        // Define a simple LED light logic

  Souliss_SetT67(memory_map, ANALOG_VALUE_SLOT2);

  pinMode(PIN_LIGHT1, OUTPUT);         // Use pin as output
  pinMode(PIN_LIGHT2, OUTPUT);         // Use pin as output

  pinMode(PIN_SWITCH1, INPUT_PULLUP);  // Hardware pulldown required
  mInput(LIGHT1_SLOT0) = 0x01;

  MaCaco_InternalSubcription(memory_map);
}

void loop()
{
  // Here we start to play
  EXECUTEFAST()
  {
    UPDATEFAST();
    // We process the logic and relevant input and output every 50 milliseconds
    FAST_110ms()
    {
      Logic_SimpleLight(LIGHT1_SLOT0);
      DigOut(PIN_LIGHT1, !Souliss_T1n_Coil, LIGHT1_SLOT0);
      DigIn(PIN_SWITCH1, Souliss_T1n_ToggleCmd, LIGHT1_SLOT0);

      Logic_SimpleLight(LIGHT2_SLOT1);
      DigOut(PIN_LIGHT2, !Souliss_T1n_Coil, LIGHT2_SLOT1);
    }
    FAST_11110ms()
    {
      Souliss_SendAnlog((float)cont, ANALOG_VALUE_SLOT2);
      cont++;
      if (cont > 1000) cont = 0;
    }

    // Here we handle here the communication with Android
    FAST_GatewayComms();
  }
}

void Souliss_SendAnlog(float value, uint8_t NUM_SLOT)
{
  Souliss_Logic_T67(memory_map, NUM_SLOT, 0.015, &data_changed);
  Souliss_ImportAnalog(memory_map, NUM_SLOT, &value);
}
```
This is my code of ESP peer 01, app, each update of android application, is changing on / off, but when you click works well, made many alteration of the code, but not solved the problem.

```ruby
// Let the IDE point to the Souliss framework
#include "SoulissFramework.h"

// Configure the framework
#include "bconf/MCU_ESP8266.h"              // Load the code directly on the ESP8266
#include "conf/IPBroadcast.h"

// **** Define the WiFi name and password ****
#define WIFICONF_INSKETCH
#define WiFi_SSID               "xxxxx"
#define WiFi_Password           "xxxxx"

// Include framework code and libraries
#include <ESP8266WiFi.h>
#include <EEPROM.h>

/*** All configuration includes should be above this line ***/
#include "Souliss.h"

// This identify the number of the LED logic
#define LIGHT1_SLOT0 0

// **** Define here the right pin for your ESP module ****
#define  PIN_LIGHT1 D4

#define ANALOG_VALUE_SLOT1 1 // Analog values use two slot, in this case 1 and 2

int cont = 0;

void setup()
{
  Initialize();

  // Connect to the WiFi network and get an address from DHCP
  GetIPAddress();

  // This is the vNet address for this node, used to communicate with other
  // nodes in your Souliss network
  SetAddress(0xAB02, 0xFF00, 0xAB01);

  Set_SimpleLight(LIGHT1_SLOT0);        // Define a simple LED light logic
  pinMode(PIN_LIGHT1, OUTPUT);         // Use pin as output

  Souliss_SetT67(memory_map, ANALOG_VALUE_SLOT1);
}

void loop()
{
  // Here we start to play
  EXECUTEFAST()
  {
    UPDATEFAST();
    FAST_50ms() // We process the logic and relevant input and output every 50 milliseconds
    {
      Logic_SimpleLight(LIGHT1_SLOT0);
      DigOut(PIN_LIGHT1, !Souliss_T1n_Coil, LIGHT1_SLOT0);
    }
    FAST_11110ms()
    {
      Souliss_SendAnlog((float)cont, ANALOG_VALUE_SLOT1);
      cont++;
      if (cont > 1000) cont = 0;
    }

    // Here we handle here the communication with Android
    FAST_PeerComms();
  }
}

void Souliss_SendAnlog(float value, uint8_t NUM_SLOT)
{
  Souliss_Logic_T67(memory_map, NUM_SLOT, 0.015, &data_changed);
  Souliss_ImportAnalog(memory_map, NUM_SLOT, &value);
}
```
I expect a light.. thank you.

plinioseniore, 2016-08-21:
Use LowDigOut instead of the ! in the parameter.

Dario.

From Mobile.

lucasmaziero, 2016-08-22:
The problem still persists..

plinioseniore, 2016-08-22:
What's the problem in detail?

From Mobile.

closed 2016-09-27 by plinioseniore