souliss/souliss #277: Control TVs & AV equipment

jkutianski, 2016-09-03:
I've a LG TV with RS232 control port, and I found some nice projects to implement IR RC & HDMI-CEC on Arduino. Maybe will be interesting add some support for TVs & AV equipment.

[image]

[LG TV RS232 control manual](https://www.lg.com/us/commercial/documents/m3701c-ba-rs232c.pdf)
[Ken Shirriff's Multi-Protocol Infrared Remote Library for the Arduino](http://www.righto.com/2009/08/multi-protocol-infrared-remote-library.html)
[Sparfun IR Communication tutorial](https://learn.sparkfun.com/tutorials/ir-communication)
[IR RC receiver](https://arduino-info.wikispaces.com/IR-RemoteControl
)
[HDMI-CEC Arduino Library](https://github.com/floe/CEC)
[Arduino Nano HDMI-CEC Board](http://www.gedasymbols.org/user/russell_dill/cec.html)

plinioseniore, 2016-09-03:
Hi Juan, welcome and thanks for your contributions.

Generally there are more people discussing on the ML, so it would be better
to post there the solution that you have applied in your home, so that you
will have a better audience.

Dario.

From Mobile.

jkutianski, 2016-09-03:
Thanks Dario.

Good point. Normally I like to see the discussions of enhancements on GH because can be tracked easily, and to me the discussion is useful to understand the changes behind the code.
I will put this on Souliss Google Group to see if someone are interested on this.
Can I leave this issue open?
Best.

Juan Pablo

plinioseniore, 2016-09-04:
Sure.

From Mobile.

closed 2017-08-16 by plinioseniore