souliss/souliss #282: string conversion errors in networksetup.cpp

bzuidgeest, 2016-09-17:
I just bought some lyt8266 bulb and I am trying to get souliss on them. so far it has not been exactly easy. Mostly troubles from finding a combination of libraries that is compatible. I have settled on: 
arduino ide 1.6.11
board package for esp8266 2.30
Latest library from souliss github download

This combo lets me build the bulb erase sample. However it does not let me build the e02_lyt8266_wifi_bulb sample. it gives many errors like this:

Documents\Arduino\libraries\souliss-friariello/base/NetworkSetup.cpp:434:39: error: cannot convert 'String' to 'const char*' for argument '1' to 'int strcmp(const char*, const char*)'

    if(strcmp(WiFi.SSID(), SSID.c_str()) || strcmp(WiFi.psk()), PSW.c_str()))

I found the file in my libraries and on each error line changed to 

Documents\Arduino\libraries\souliss-friariello/base/NetworkSetup.cpp:434:39: error: cannot convert 'String' to 'const char*' for argument '1' to 'int strcmp(const char*, const char*)'

    if(strcmp(WiFi.SSID().c_str(), SSID.c_str()) || strcmp(WiFi.psk().c_str(), PSW.c_str()))

I do not know if that is the best way to go around fixing the problem as my cpp knowledge is limited.

I thought I would place it here.

juanpintom, 2016-09-18:
Try with 1.6.8 IDE and 2.2.0 Core :P

Read at bottom here:
https://github.com/souliss/souliss/wiki/Supported%20Hardware

Thanks

bzuidgeest, 2016-09-19:
Sorry, but I'm not interested in trying the old versions of the ide and esp core packages. I'm sure the version combination you suggest works fine. But I use the latest versions in my projects, your ide choice is three versions out of date and thus lacks the many fixes made in the newer versions.

My primary reason for opening this issue is notifying you of the problem. My suggested changes works for me and allow met to run souliss with the newer versions of the ide and board packages. The performance was pretty great.

plinioseniore, 2016-09-27:
@bzuidgeest that fix is included in Souliss 7.2.0 that is available in 720_rc branch, that code is not yet released and so is not available in the Board Manager.

closed 2016-09-27 by plinioseniore