souliss/souliss #291: Output status T22

manuservel, 2017-04-14:
Hello,
Using the T22 I get output status 16(hex) for open. 
In https://github.com/souliss/souliss/wiki/Typical_T2n says:
Output status:
    1(hex) for CLOSING,
    2(hex) for OPENING,
    3(hex) for STOP.
    8(hex) for CLOSE
    10(hex for OPEN
Why do I get 16 and not 10 for OPEN?
Thank you.

plinioseniore, 2017-04-14:
This looks the reason for that https://github.com/souliss/souliss/blob/friariello/Typicals.h#L164, there is no much sense in 0x16 and should be changed as 0x10.

plinioseniore, 2017-04-14:
@FulvioSpelta any comment on this?

plinioseniore, 2017-04-14:
@shineangelic witch value is used in SoulissApp for OpenLimitSwitch state?

manuservel, 2017-04-14:
Hello,
I do not use limit switch. All other states match their hexadecimal value.

FulvioSpelta, 2017-04-15:
Hi the line for the state values is: 
[https://github.com/souliss/souliss/blob/friariello/Typicals.h#L171]that is 0x10

Pls post your sketch the problems seems unclear to me.
thx

plinioseniore, 2017-04-15:
0x16 is LimitSwitchOpen so when a limit switch is reached this state is
activated.

My previous post was wrong as 0x10 is used for Open state (without limit
switch).

As you are not using limit switches, you should never reach 0x16.

Dario.

From Mobile.

FulvioSpelta, 2017-04-15:
Hi all
the constants `Souliss_T2n_LimSwitch_Close` and `Souliss_T2n_LimSwitch_Open` are used as "input only" and shouldn't never go to the output.

I've found an error in line 126 that currently is:
`memory_map[MaCaco_OUT_s + slot] = Souliss_T2n_LimSwitch_Open;			// Open Limit Switch`
it should be:
`memory_map[MaCaco_OUT_s + slot] = Souliss_T2n_State_Open;			// Open Limit Switch`
but its position in the code doesn't explain the @manuservel error. I kindly ask @manuservel to modify its souliss library and check if this correction fix his problem so I can commit the change.
Thanks

plinioseniore, 2017-04-16:
That make sense, but I guess that also SoulissApp use LimitSwitch as both
input and output value.

From Mobile.

FulvioSpelta, 2017-04-16:
Hi Dario
a full review is needed (eg: in T21 comments the constants are wrong) and we have to review:
- All T2x code and comments (I've already get a look and it should be ok excluding the comments of the T21)
- SoulissApp assumption
- Openhab binding assumption (translation table)

I'll do it in the next days, in the meanwhile I'd like to have a feedback on testing because I'm in trouble to test the code in real environment.

I'll post here the further reviews.

shineangelic, 2017-05-08:
If the typicals' comments are wrong FOR SURE there are problem in the app as well

plinioseniore, 2020-01-06:
@manuservel know that this is from long time ago, but had you the opportunity to test the solution proposed by @FulvioSpelta ?