souliss/souliss #309: In some situations a node will retry the join process before the gateway has time to respond

alvarolobato, 2017-10-17:
With some configurations and number of nodes, a new joining node may be too aggressive trying to join and will aggressively retry to join, causing the join process never end because the gateway will try to refresh all the nodes before adding the new one.

Adding some more time to the retry process will do in most of the cases, but the process seems to be fragile and may require a redesign.

This is an example where AB0D is trying to join:

Node Logs
```
Init Souliss...(vNet)<USERMODE><0x1E06|0x4786|0x55CA|0x10AE|0x3EAE|0x>
OK
Init relays...OK
Init PINS...OK
Get IP Address...(vNet)<SETADDR><|0x77>
(MAC)<1A,A6,49,6B,0,78,>
(IP)<C0,A8,1,77,>
192.168.1.119
Set as Gateway...OK
Init addressing server...(vNet)<SETADDR><|0xAB01>
(ss)<rPddr><|0x77|0xAB05|0xCE02|0xAB02|0xCE03|0xAB04|0xAB06|0xAB07|0xAB08|0xAB09|0xAB0A|0xAB0B|0xAB0C|0xAB0D|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x>
OK
Got Address: AB01
Other addresses:
 1:77
 2:0
 3:AB01
 4:0
 5:0
Initialization finished
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB05><|0xB|0x17|0xAB05|0xAB01>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x5>
(vNet)<NHBOR><|0x5>
(vNet)<DONTROUTE>
(vNet)<MEDIA><|0x5>
(vNet)<OUT><DADDR><|0x0><|0xB|0x17|0xCE02|0x0>
(vNet)<OUT>-FAIL
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB02><|0xB|0x17|0xAB02|0xAB01>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x5>
(vNet)<NHBOR><|0x5>
(vNet)<DONTROUTE>
(vNet)<MEDIA><|0x5>
(vNet)<OUT><DADDR><|0x0><|0xB|0x17|0xCE03|0x0>
(vNet)<OUT>-FAIL
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB04><|0xB|0x17|0xAB04|0xAB01>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB06><|0xB|0x17|0xAB06|0xAB01>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB07><|0xB|0x17|0xAB07|0xAB01>
(vNet)<IN>(0xAB07)<|0x23|0x17|0xAB01|0xAB07|0x15|0x0|0x0|0x57|0x18|0x0|0x0|0x1|0x0|0x0|0x0|0xFE|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(MaCaco)<IN><|0x15|0x0|0x0|0x57|0x18|0x>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
(vNet)<MEDIA><|0x3>
(vNet)<OUT><DADDR><|0xAB08><|0xB|0x17|0xAB08|0xAB01>
(MaCaco)<OUT><|0x5|0x0|0x0|0x57|0x18|0x>
(vNet)<OUTPATH><|0x3>
[... 236 lines cut ...]
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB06|0xAB01|0x5|0x0|0x0|0x57|0x18>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB08|0xAB01|0x5|0x0|0x0|0x57|0x18>
REQUEST JOIN
(MaCaco)<OUT><|0x2C|0x0|0x0|0x0|0x0|0x>
(vNet)<BRD><Media><|0x1><|0xB|0x17|0xFFFF|0x7E>
(vNet)<BRD><Media><|0x3><|0xB|0x17|0xFFFF|0xAB0D>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB02|0xAB01|0x5|0x0|0x0|0x57|0x18>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB04|0xAB01|0x5|0x0|0x0|0x57|0x18>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB06|0xAB01|0x5|0x0|0x0|0x57|0x18>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB07|0xAB01|0x5|0x0|0x0|0x57|0x18>
(vNet)<IN>(0xAB07)<|0x23|0x17|0xAB01|0xAB07|0x15|0x0|0x0|0x57|0x18|0x0|0x1|0x0|0x0|0x0|0x0|0xFE|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0|0x0>
(vNet)<IN>(0xAB01)<|0xB|0x17|0xAB09|0xAB01|0x5|0x0|0x0|0x57|0x18>
REQUEST JOIN
(MaCaco)<OUT><|0x2C|0x0|0x0|0x0|0x0|0x>
(vNet)<BRD><Media><|0x1><|0xB|0x17|0xFFFF|0x7E>
(vNet)<BRD><Media><|0x3><|0xB|0x17|0xFFFF|0xAB0D>
(vNet)<IN>(0xAB09)<|0xB|0x17|0xFFFF|0xAB09|0x2B|0x0|0x0|0x0|0x0>
```

In this case increasing the number of retries here:
https://github.com/souliss/souliss/blob/63c3212ffe45b89ba202f9236d12cd42d783388a/base/SpeakEasy.h#L397
and here:
https://github.com/souliss/souliss/blob/63c3212ffe45b89ba202f9236d12cd42d783388a/base/SpeakEasy.h#L404

did the trick.

plinioseniore, 2020-01-06:
Hi @alvarolobato this is from long time ago, I saw in the code that retries has been increased from 100 to 1000, what's the value you are using?

alvarolobato, 2020-01-08:
Hi, I switched to static addressing now, so this is not a problem for me anymore.

closed 2022-11-23 by alvarolobato