souliss/souliss #325: Add DigInHold2 and LowDigInHold2

sdurnev, 2022-03-27:
Hey! 
I suggest adding the following features.

### DigInHold2

Similar to DigInHold, but controls two different slots.

Single press ->  value_state1 -> slot1

Long press ->  value_state2 -> slot2

```C
U8 DigInHold2(U8 pin, U8 value_state1, U8 value_state2, U8 slot1, U8 slot2);
```

plinioseniore, 2022-03-27:
Hi, can you please submit a pull request?

sdurnev, 2022-03-27:
Can't create a new pull request: Push failed: remote: Permission to souliss/souliss.git denied to sdurnev. unable to access 'https://github.com/souliss/souliss.git/': The requested URL returned error: 403

Maybe I'm doing something wrong?

plinioseniore, 2022-03-27:
I see you make two direct commits, one adding your code and one deleting

https://github.com/souliss/souliss/commit/f0b07fbd6dac884653bc949a78c0b63c04174e7e

https://github.com/souliss/souliss/commit/fe3b239c045bf4f254a397eac0066830691e4e5f

sdurnev, 2022-03-27:
You're right.
I don't have much experience unfortunately.
This time my code is added.
I will consider your recommendations in the future.
Thanks!

closed 2022-03-28 by sdurnev